atc1441/ATC_MiThermometer #78: Add custom firmware builds with various default settings

ptmouly, 2020-11-14:
Could you add to your repo other bin files with alternate default settings.
I don't have the compilation env to build it myself and I'd like to have custom firmware with mi-like advertisement by default.
Otherwise each time I change the battery of my  item I have to change the settings again...

Thanks and kudos for the good work !

atc1441, 2020-11-14:
The big problem with that is that every update needs to be made to every file. Thats just to much.

In the future there will be persistence storage so you only need to set it once

ptmouly, 2020-11-14:
Oh ok I see. Thanks for your quick answer. I'm waiting for the next version then !

jperquin, 2020-12-25:
Has persistent storage been implemented?

atc1441, 2020-12-25:
Unfortunately still no.

Being very ocupied by another project that involves my full free time