atc1441/ATC_MiThermometer #80: cstartup_8251 or cstartup_8258?

pvvx, 2020-11-14:
AN_19011501-C5_Telink Kite BLE SDK Developer Handbook.pdf
https://yadi.sk/i/FsMuapzANxT78Q
[image]

cstartup_8258:
```
    ...
	tmov	r0, #0x13		@r0 = 0x13 SVC
	tmcsr	r0				@CPSR = r0
	tloadr	r0, DAT0 + 12	@r0 = __RAM_START_ADDR + __RAM_SIZE_MAX
	tmov	r13, r0  		@r13 = r0 update SP under SVC mode
	...

DAT0:
	.word	0x12			    @IRQ    @0
	.word	0x13			    @SVC    @4
	.word	(irq_stk + IRQ_STK_SIZE)
	.word	(**0x850000**)		    @12  stack end 
```
stack end -> 64k!

?
https://github.com/atc1441/ATC_MiThermometer/blob/master/ATC_Thermometer/makefile#L45
#ifeq ($(RETENTION_RAM_SIZE), 32KB)
	LS_FLAGS := $(TEL_PATH)/components/boot/boot_32k_retn_8253_8258.link
#else
#	LS_FLAGS := $(TEL_PATH)/components/boot/boot_16k_retn_8251_8253_8258.link
#endif

SDK Ai-Thinker:
BOOT_FLAG := -DMCU_STARTUP_8258_RET_16K
https://github.com/Ai-Thinker-Open/Telink_825X_SDK/blob/master/make/boot.mk#L19

?
BOOT_FLAG := -DMCU_STARTUP_8251_RET_16K
https://github.com/Ai-Thinker-Open/Telink_825X_SDK/blob/master/components/boot/8251/cstartup_8251_RET_16K.S#L23

pvvx, 2020-11-14:
I tested TLSR8253F512ET32 (Ai-Thinker BT-01..04 modules).
https://docs.ai-thinker.com/blue_tooth
All have 64 kilobytes of SRAM. Inside the crystal 8258 (?)

pvvx, 2020-11-14:
ATC_MiThermometer uses a large 'Wasteful SRAM Area' and in firmware.
Real end 'retention data' address 0x4400. Aligned(256).
0x4400..0x8000 = 'Wasteful SRAM Area' and data into Flash.

Reduce firmware size for flash and load speed.
Set IC-cache start address 0x4400.
[cstartup_825x.zip](https://github.com/atc1441/ATC_MiThermometer/files/5541548/cstartup_825x.zip)

atc1441, 2020-11-14:
Thank you for that.

It is a bit above my head right now as i am fully into another topic but will ger back to it

pvvx, 2020-11-15:
Building Reduced firmware size
https://github.com/pvvx/ATC_MiThermometer

closed 2020-11-15 by pvvx