atc1441/ATC_MiThermometer #81: Device crashing/rebooting

TheNitek, 2020-11-15:
Currently I have two thermometers running this firmware. While one is running fine, the other one seems to restart ~once a day (= back to default settings). According to the display the battery is still >50%, so I would not expect this to be the issue (.
Has anyone else seen this?

atc1441, 2020-11-15:
Can you try to reflash that one. 

It could be a byte error in the flash

TheNitek, 2020-11-15:
I already did (and updated to the latest firmware version, so it's also not tied to a specific firmware version)

atc1441, 2020-11-15:
Mhh maybe the battery is very low and the measured percent is only the floating voltage and under load it goes down.
But thats just an idea

TheNitek, 2020-11-15:
Probably you are right - I was just hoping to get around switching the battery 😄 Looks I have to do it anyway.

closed 2020-11-15 by TheNitek

atc1441, 2020-11-15:
Its good to get some feedback about this issue, 
mine still run fine so i never tested a dead battery 

Hipe the battery change resolves the issue

TheNitek, 2020-11-15:
I was just about the switch battery between both device to see if the issue switches as well, when I saw the reboot including a display a could barely read (low contrast), so I am quite sure it's battery.