atc1441/ATC_MiThermometer #82: I cannot activate my device

onlize, 2020-11-17:
Hello, I cannot activate my device. It tells me either:
23:03:11: Activating now, please wait...
23:03:12: Update error: some error while sending char data
23:03:12: Update error: some error while sending char data

Or:
23:03:19: Activating now, please wait...

Here is the picture:
[image]

I try to make it work for a couple of hours now. I tried Google Chrome, Firefox, and Microsoft Edge. Can you help?

atc1441, 2020-11-17:
Hey.

Could you also share the console log from chrome?
( at debug tools ) that could help to get further

onlize, 2020-11-17:
Thank you for your reply. I am not sure if this is what you are looking for, but this is the only place for logs that I see:

[image]

atc1441, 2020-11-17:
Hey  that is not what meant but looks a bit strange. Try restarting both pc and device

onlize, 2020-11-17:
Restarting everything helped. Thank you.

closed 2020-11-17 by onlize