atc1441/ATC_MiThermometer #84: Add license information

vazhnov, 2020-11-17:
Hello!

I found link to this cool project from https://hackaday.com/2020/11/17/custom-firmware-for-cheap-bluetooth-thermometers/

Could you please add information about license of this repository — is it allowed to copy, change and use code in any purpose?
It would be great to see any popular open source license.

vazhnov, 2020-12-08:
@atc1441 ?

atc1441, 2020-12-08:
Hey. Sorry

I thought about what license to choose but totally forgot about it.

I am still not shure, most likely it will be GPLv3 as its the best for all and i dought anyone will use this commercially

vazhnov, 2020-12-08:
@atc1441 , license choose is not easy, of course. You can set one license (GPL v3 for example) and mention that if somebody needs more libre license (MIT/Apache/BSD), then ask in project issues.

vazhnov, 2020-12-08:
Anyway, any popular opensource license is much better than no license (proprietary code).

atc1441, 2020-12-10:
I havent forgot you! 

Will make it GPLv3 :)