atc1441/ATC_MiThermometer #87: Question: What about possible support for the LYWSD02MMC ?

markbirss, 2020-11-18:
Hi

This is just a question about possible support for the LYWSD02MMC ? 

Thank you for your effort on the LYWSD03MMC i hope to receive some devices soon to try.

atc1441, 2020-11-18:
Hey. 

The problen is that it uses a different SOC so is in itself a complete new project to do.

It was allready opened to see how easy it would be.

The most likely next device is the MHO-C401
As it also uses the TLSR8251

markbirss, 2020-11-18:
Ah Ok, understood, thank you for your answer and feedback

closed 2020-11-18 by markbirss

TheGroundZero, 2021-02-04:
Oh, so the LYWSD02MMC isn't supported?
Darn it ... already bought one :/

ceinmart, 2023-11-26:
Hi @atc1441 , @pvvx
Can we have hope for LYWSD02MMC ? 
I bought one that I was believed could be used without the need a custom firmware, but it comes with fw v2 . 
Apparently, it works well with fw v1 only... 

Really need a way to downgrade or a great custom fw like D02 model...

pvvx, 2023-11-26:
LYWSD02MMC - https://pvvx.github.io/LYWSD02/ 
SoC: DA14585