atc1441/ATC_MiThermometer #95: Feature request: Add 2 point calibration would improve accuracy a lot

JsBergbau, 2020-11-20:
Hallo @atc1441,

thank you very much for this custom firmware. I use these sensors for almost one year now and have quite some experience with them.
With your custom firmware humidity values are displayed about 5 percent points lower than with original firmware. I've read that you take the direct reading of the sensor. Original firmware seems to process this value.

 I'm now in the process of calibrating with your custom firmware flashed devices. I calibrate them with MgCl (33 % r.H.) and NaCl (75 % r.h.). With MgCl readings are with most sensors 34 % and 35 %, so they fit quite well. However with NaCl readings are most 70 %, some 71 % whereas original firmware shows here 75 %. It seems for me Xiaomi introduced this offset to make them more accurate in the area of higher humidities, whereas with your firmware accuracy is higher in the lower areas.

These sensors seem to be not very linear which seems quite normal:
https://www.rotronic.com/media/productattachments/files/c/a/capacitive_humidity_sensor_final.pdf
[image]

So to have higher accuracy we can implement a 2 point calibration. Perhaps we can even use default values based on my calibration experiences (in total I will get the values for 20 - 30 LYWSD03MMC sensors). But for my experiences these values depent a lot on the batch. My first batch is quite different from the last batch concerning calibration values.

Indepent of default values I would find it a great idea to have a 2 point calibration in your firmware. Then the readings on the display are quite accurate and not only the processed values.

You can adopt my python function for calibrating humidity with 2 points:

```
def calibrateHumidity2Points(humidity, offset1, offset2, calpoint1, calpoint2):
	p1y=calpoint1
	p2y=calpoint2
	p1x=p1y - offset1
	p2x=p2y - offset2
	m = (p1y - p2y) * 1.0 / (p1x - p2x) # y=mx+b
	b = p2y - m * p2x #would be more efficient to do this calculations only once
	humidityCalibrated=m*humidity + b
	if (humidityCalibrated > 100 ): #with correct calibration this should not happen
		humidityCalibrated = 100
	elif (humidityCalibrated < 0):
		humidityCalibrated = 0
	humidityCalibrated=int(round(humidityCalibrated,0))
	return humidityCalibrated
```

For further infos, please take a look at https://github.com/JsBergbau/MiTemperature2#calibration
If you want to try yourself: There are a lot of MgCl offers, please take one which is very pure.
I think you're from Germany. Here https://www.ebay.de/itm/2-kg-EUR6-45-kg-Magnesiumchlorid-Hexahydrat-MgCl2-99-reinst-Pharma-Neu/253094314322 you can get it very pure and with a good price-performance ratio.

atc1441, 2020-11-20:
Thank you, will put it on my list of Todo, the persistence storage is definitely first.

but i understand the need of calibration as well :) 

Xiaomi has some 5% offset but for me it looked like they made a general offset and not a 2 point calibration, never looked deeper into that.
The values this firmware uses are the direct ones from the sensor without any filter

JsBergbau, 2020-11-20:
Thank you very much for putting it onto your todo list. 

Yeah I also think Xiomi just uses an offset calibration. Sorry if my post was misleading. Its just me / the script that uses/offers 2 point calibration.

mredone, 2020-11-25:
It will be great adopt 2 point calibration to temp too. 
This thermometers is weak. To 21 degree it read -0,6 degree lower than sample thermometer. But at 22 degree it read same temp like sample thermometer.

Ps. English is my 3rd language.

JsBergbau, 2020-11-25:
How stable was the temperature and how long did give the Thermometers time to show the same temperature? Either your Xiamomi LYWSD03MMC Thermometer was faulty or it was the measuring conditions. I can't imagine that it shows 0,6 degress difference whithin 1 degree temperature difference. Especially at these low temperature differences it can take very long to show the same temperature and thus it is very important to have a very stable temperature over that long time. 

Currently I'm calibrating humidity of 15 of these LYWSD03MMC devices. All show currently 20,4 or 20,5 °C. Whereas the humidity readings differs a lot more. Will report back when calibration of all 30 devices is completed. It takes at least 30 hours to get a constant humidity inside my calibration box of only 25 cm x 15 x 5 cm dimension and also there is realtive constant temperature is required over the calibration process.

mredone, 2020-11-25:
I have six LYWSD03MMC devices. All six have the same problem. One is -0.6, second -0.4, rest -0,3 but when temperature is above 22 tempereture this "error" disappears.
He compares them to the commercial Salus thermostat, the indications of which correspond to other thermometers, e.g. from xiaomi WSDCGQ11LM.

atc1441, 2020-11-25:
Hey. 

Yeah the sensor is very sensitive.

Even if you just take the thermometer in the hand the temperature will change a lot and takes time to aclimate

tetsuo55, 2021-08-18:
> How stable was the temperature and how long did give the Thermometers time to show the same temperature? Either your Xiamomi LYWSD03MMC Thermometer was faulty or it was the measuring conditions. I [...]

Did you finish the calibration? Is there an offset value that those of us who cannot calibrate can use?

JsBergbau, 2021-08-19:
Yes I've finished calibration. However it seemed it depended on the batch of the device how much calibration was needed. So there is no value that I can recommend for all devices. You really have to do it by yourself, if you really like very accurate readings. However the devices are even without calibration pretty exact with humidity values.

tmugan, 2023-11-28:
Was there some outcome for this about getting 2 points of calibration?
I've just purchased a dehumidifier and the RH values of my LYWSD03MMC B1.6 sensors are quite different (flashed with v4.5 recently). 
I only see a single Temp and Humidity offset value available in the config of those sensors.