atc1441/ATC_MiThermometer #97: Transmitting sub-zero (negative) temperatures

Einstein67, 2020-11-21:
Hello and thanks for this great software.

I know the specification of this sensor is 0-60 degrees. But the sensor also measures negative values without any problems and shows them correctly in the display. However, incorrect values are then sent.

Stock firmware sends a value of "+ 654.78 ° C" at "-1.6 ° C"
ATC firmware sends "+ 6532.2 ° C" for the same measured value

Can the measuring range be extended to -9.9 ° C (Limitation of the characters in the display)? It seems that the sensor could, but the communication isn't designed for it.

Or am I making a mistake?

Thanx

atc1441, 2020-11-21:
Hey, 

that sounds like you are parsing the Temp as an UINT16_T and not as an INT16_T as it is intened.

When negative values are send the first bit is 1 which indicates negative with an INT value but it represends a very high number at an UINT

hope that helps

Einstein67, 2020-11-21:
Ok, then that needs to be fixed in Tasmota.

Incidentally, "Telink flasher" also shows a wrong value with original software.

[image]

Thanks

closed 2020-11-21 by Einstein67

atc1441, 2020-11-21:
Good catch with the telink flasher. 

Will check on that.

atc1441, 2020-11-21:
Fixed it in the newest WebFlasher

Einstein67, 2020-11-22:
TASMOTA will also be fixed soon

[image]

Thanks again

ralf-e, 2020-11-22:
> TASMOTA will also be fixed soon

how did you configure tasmota for ATC, which esp32 binary for BLE?

Einstein67, 2020-11-22:
I compiled with "#define USE_MI_ESP32" and activated BLE with "SetOption115 1".

ralf-e, 2020-11-22:
OK, thanks, that's how it works.

Do you know how many BLE sensors Tasmota can correctly query.
I have just run a test with 12 sensors (ATC and one LYWSD02).
All 12 are displayed in the Tasmota web interface, but only the first 10 of them are sent to the MQTT server.