atc1441/ATC_MiThermometer #98: No data in HomeAssistant

joshuagti, 2020-11-22:
Trying here to see if anyone has any insight! Using an ESP32.

Firstly I used the tool to get the bindkey to enter into Esphome and although it would see the device and using the generated bindkey it would not show any sensor data.

I then though to flash the firmware and it appears to have gone okay and flashes my MAC address on boot and the device name showing in the esp32 logs but still no sensor data (shows "unknown").

```
esp32_ble_tracker:
sensor:
  - platform: xiaomi_lywsd03mmc
    mac_address: "A4:C1:38:B7:F4:D7"
    bindkey: "ab1a981ed5b3ac61e0aac620b8c1939a"
    temperature:
      name: "Bedroom Temperature"
    humidity:
      name: "Bedroom Humidity"
    battery_level:
      name: "Bedroom Temp Sensor Battery Level"
```

Before I flashed I could see the temp data in the tool but now it shows "waiting for data, this will not change on devices with custom firmware" - unsure whether this is supposed to be like this? No device known ID, Mi token or Mi bindkey is also showing now.

atc1441, 2020-11-22:
Hey did you set the data to Mi Like via the WebFlasher after flashing the custom firmware?

joshuagti, 2020-11-22:
Thanks for the quick reply, I pressed it like 20 times originally and after I opened the issue, I did the old fashioned turn everything off and on again (last ditch effort), connected to it on my phone, pressed the button about 50 times, disconnected bluetooth on my phone, pulled battery on sensor and then the data appeared - hoorah!

Not sure exactly what part fixed my issue but left it all in there in case anybody else has similar issue

closed 2020-11-22 by joshuagti

atc1441, 2020-11-22:
Good to know it works now. 
Even if the reason is unknown :)

Zixim, 2020-12-25:
I also ran into this.
Isn't there a way of verifying the "Mi Like" setting in the sensor from the Flasher page ?

atc1441, 2020-12-25:
You can verfiy it with the app nRFconnect, lool at what uuid is advertising tze data if it is 0x95FE then its "MiLike"