atc1441/ATC_MiThermometer #100: Can't figure connect to 4th unit

JABIoT, 2020-11-25:
I am having an issue with the 4th temp sensor.
I have HA 0.118.3
ESPHome 1.15.3

I had 4 xiaomi working well for a couple of months. Then I put one in a freezer and it stopped connecting to ESPHome although I still can see it on the atc1441 web tool, tried to reflash the bin file and that took about 30 minutes so I figured something was wrong with the xiaomi module.

I purchased more modules, flashed the bin file and modified ESPHome to reflect the new MAC and uploaded the new build to ESP32. looking at the ESP32 log, it will not detect the 4th module. I waited a few hours and still not detected. 
What else can I try?

This is the ESPHome code:

`substitutions:
  esphome_name: nodetemp

esphome:
  name: ${esphome_name}
  platform: ESP32
  board: mhetesp32devkit

wifi:
  ssid: "****"
  password: "****"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Nodetemp Fallback Hotspot"
    password: "****"

captive_portal:

# Enable logging
logger:
  level: debug

# Enable Home Assistant API
api:
  reboot_timeout: 60min
  password: "****"

ota:
  password: "****"

# Enable Bluetooth scanning for this ESP32
esp32_ble_tracker:

time:
  - platform: homeassistant
    on_time:
      - seconds: 0
        minutes: 0
        hours: 4
        days_of_week: MON-SUN
        then:
           - switch.toggle: restart_switch
switch:
  - platform: gpio
    name: "${esphome_name}-Onboard-LED"
    pin: 2
    inverted: True
  - platform: restart
    name: ${esphome_name}-restart
    id: restart_switch
  
sensor:  
  - platform: uptime
    name: "${esphome_name}_Uptime Sensor"
    
  - platform: wifi_signal
    name: "${esphome_name} WiFi Signal"
    update_interval: 60s
    
  - platform: xiaomi_lywsd03mmc
    bindkey: "3a8e210a28b9d0b6d8aaf8862e9c81a3"
    mac_address: A4:C1:38:xx:xx:xx
    temperature:
      name: "Tempunit1"
    humidity:
      name: "Humidunit1"
    battery_level:
      name: "Battunit1"

  - platform: xiaomi_lywsd03mmc
    bindkey: "5b27cf5b4afc233956982d1114ec2bb7"
    mac_address: A4:C1:38:xx:xx:xx
    temperature:
      name: "Tempunit2"
    humidity:
      name: "Humidunit2"
    battery_level:
      name: "Battunit2"

  - platform: xiaomi_lywsd03mmc
    bindkey: "3a8e210a28b9d0b6d8aaf8862e9c81a3"
    mac_address: A4:C1:38:xx:xx:xx
    temperature:
      name: "Tempunit3"
    humidity:
      name: "Humidunit3"
    battery_level:
      name: "Battunit3"

  - platform: xiaomi_lywsd03mmc
    bindkey: "3a8e210a28b9d0b6d8aaf8862e9c81a3"
    mac_address: A4:C1:38:xx:xx:xx
    temperature:
      name: "Tempunit4"
    humidity:
      name: "Humidunit4"
    battery_level:
      name: "Battunit4"`

atc1441, 2020-11-25:
Hey unfortunately I think I cannot help you with it as I don't use ESPhome

You could try to restart and reflash it. Maybe that will help

JABIoT, 2020-11-25:
I have already tried to restart the ESP32, pulled the battery out of the xiaomi and restarted HA but no luck.
The only thing I am unable to validate is the MAC starting with A4:C1:38 is it possible the new module I purchased has something different?

michapr, 2020-11-25:
what do you see in your smartphone while scanning Bluetooth network?
Here must be visible any of the devices - with name.

JABIoT, 2020-11-25:
My phone did not detect this device when scanning for bluetooth.
I found an APP to scan Bluetooth devices and it found it and it does start with A4:C1:38.

Looking at the log from ESPhome it is detecting it and showing only this info"

`[14:26:18][D][esp32_ble_tracker:544]: Found device A4:C1:38:xx:xx:xx RSSI=-57
[14:26:18][D][esp32_ble_tracker:565]:   Address Type: PUBLIC
[14:26:18][D][esp32_ble_tracker:567]:   Name: 'ATC_xxxxxx'`

But it never get to show this like it does for other devices:

`[14:21:32][D][xiaomi_ble:300]: Got Xiaomi LYWSD03MMC (A4:C1:38:xx:xx:xx):
[14:21:32][D][xiaomi_ble:303]:   Temperature: 20.5°C
[14:21:32][D][xiaomi_ble:306]:   Humidity: 32.0%
[14:21:32][D][sensor:092]: 'Tempunit3': Sending state 20.50000 °C with 1 decimals of accuracy
[14:21:32][D][sensor:092]: 'Humidunit3': Sending state 32.00000 % with 0 decimals of accuracy`

JABIoT, 2020-11-25:
I believe I have 2 issues.
1 - I blew the first unit by putting it in the freezer.
2 - I am under the impression the ATC bin file is no longer working.

I took a new xiaomi module and just got the blind key from the Web tool, updated the MAC in ESPhome and I was able to succesfully read the temperature and humidity in HA but not the battery level. Knowing this module was working, I flashed with the bin file, reconnected and selected the following options:
Smily: off
Advertising Type: Mi Like
Show battery in LCD: Disabled
Advertising int. 5 Min

Looking at the Web tool log, all changes where succesful.

Now, I no longer have any readings from the unit.

atc1441, 2020-11-26:
You could try to flash the devices back to the stock firmware and see if sey work like they should 

Also you could try an older version of ATC bin file ( see the revisions here on github )

JABIoT, 2020-11-26:
I tried the bin file that was working when I first setup the 4 modules and that did not work.
Flashing back to stock also did not work.
I tried connecting with the App provided by xiaomi to the module that was flashed back to stock and I got this:
[image]

But connecting with the xiaomi App only if I select China as the country works on an unflashed module.
Could it be that xiaomi changed something in their modules?

JABIoT, 2020-11-26:
I am really confused but it started working.
After connecting with the Web tool and re-selecting
Smily: off
Advertising Type: Mi Like
Show battery in LCD: Disabled

it is indicated that those settings are lost after removing power. At Advertising type, if My Like is not selected will it stop working?
That would explain why I could not get it to work, maybe i didn't wait long enough and then I reset by removing the battery.

atc1441, 2020-11-26:
Good to know it is working now,

Could je that error you describe.

To the mi activation, flashing of the custom firmware should remove any custom changes but you are not allowed to activate it via webflasher and then add it to mi home they will notice it then and have blocked it :D 
So only flashing stock and then use mi home

closed 2020-12-07 by atc1441