atc1441/ATC_MiThermometer #101: Any luck you can share the Mi Authentication Code

imbuggy, 2020-11-27:
1/ Would be nice to add support for other Mi devices that require bind key. Would you mind sharing the code for it so we can contribute? 

You reference this 'https://github.com/danielkucera/mi-standardauth/blob/master/provision.py' in the thank you notes but the author says it is not functional, were you able to get it to work?

2/ Alternatively, Currently the web app disconnects because it can't find the 00010203-0405-0607-0809-0a0b0c0d1912 service (OTA). Any luck we could make this optional so that we can still do the activation with the 0xFE95 service on other Xiaomi devices and get a bind_key?

Thank you in advance.

imbuggy, 2020-11-27:
For example with the MCCGQ02HL (Smart Door Window Sensor 2 With Light Detection):

Log:
18:42:11: Searching for devices
18:42:26: Connecting to: standard demo
18:42:30: NotFoundError: No Services matching UUID 00010203-0405-0607-0809-0a0b0c0d1912 found in Device.
18:42:30: Reconnect 1 from 5
18:42:30: NotFoundError: No Services matching UUID 00010203-0405-0607-0809-0a0b0c0d1912 found in Device.
18:42:30: Reconnect 2 from 5
18:42:30: NotFoundError: No Services matching UUID 00010203-0405-0607-0809-0a0b0c0d1912 found in Device.
18:42:30: Reconnect 3 from 5
18:42:30: NotFoundError: No Services matching UUID 00010203-0405-0607-0809-0a0b0c0d1912 found in Device.
18:42:30: Reconnect 4 from 5
18:42:30: NotFoundError: No Services matching UUID 00010203-0405-0607-0809-0a0b0c0d1912 found in Device.
18:42:30: Reconnect 5 from 5
18:42:30: NotFoundError: No Services matching UUID 00010203-0405-0607-0809-0a0b0c0d1912 found in Device.
18:42:30: Something went wrong, to many reconnect's
18:48:38: Disconnected.

Services:
[image]

atc1441, 2020-11-27:
Please try out this version: https://atc1441.github.io/Temp_universal_mi_activate.html

do it on your own risk :D 

this is just a quick hack with everything else deactivated except the activation itself.

i can not real share any manual on how the activation works as the .html file is the only info i got myself it was developed until it worked. 

did not got the python script to work but it was a big help

imbuggy, 2020-11-27:
Hi, thank you for your swift reply.

I tried with two different Xiaomi services (MCCGQ02HL and MJYD2S). It looks like it didnt get very far in the activation. Last notification is (0x) 01-00-01-00-00-00 on attribue 19.

Log:
20:20:12: Searching for devices
20:20:14: Connecting to: standard demo
20:20:14: Detected Mi Thermometer
20:20:15: Connected
20:20:27: Activating now, please wait...
20:23:50: Disconnected.
20:23:50: Searching for devices
20:24:03: Connecting to: MJYD2S
20:24:04: Detected Mi Thermometer
20:24:04: Connected
20:24:18: Activating now, please wait...

Partial BLE Log from second connection:

nRF Connect, 2020-11-27
MJYD2S (50:EC:50:E1:EC:B0)
D	20:24:09.604	[Broadcast] Action received: android.bluetooth.device.action.ACL_CONNECTED
V	20:24:09.855	Connecting to 50:EC:50:E1:EC:B0...
D	20:24:09.855	gatt = device.connectGatt(autoConnect = false, TRANSPORT_LE, opportunistic = true, preferred PHY = LE 1M) (hidden)
D	20:24:09.869	[Callback] Connection state changed with status: 0 and new state: CONNECTED (2)
I	20:24:09.869	Connected to 50:EC:50:E1:EC:B0
V	20:24:09.883	Discovering services...
D	20:24:09.883	gatt.discoverServices()
D	20:24:09.891	[Callback] Services discovered with status: 0
I	20:24:09.891	Services discovered
V	20:24:09.908	Generic Attribute (0x1801)
- Service Changed [I] (0x2A05)
   Client Characteristic Configuration (0x2902)
Generic Access (0x1800)
- Device Name [R] (0x2A00)
- Appearance [R] (0x2A01)
- Peripheral Preferred Connection Parameters [R] (0x2A04)
- Central Address Resolution [R] (0x2AA6)
Battery Service (0x180F)
- Battery Level [N R] (0x2A19)
   Client Characteristic Configuration (0x2902)
Device Information (0x180A)
- Manufacturer Name String [R] (0x2A29)
- Model Number String [R] (0x2A24)
- Hardware Revision String [R] (0x2A27)
- Firmware Revision String [R] (0x2A26)
Unknown Service (0000fe95-0000-1000-8000-00805f9b34fb)
- Unknown Characteristic [R] (00000004-0000-1000-8000-00805f9b34fb)
- Unknown Characteristic [N WNR] (00000010-0000-1000-8000-00805f9b34fb)
   Client Characteristic Configuration (0x2902)
- Unknown Characteristic [N WNR] (00000019-0000-1000-8000-00805f9b34fb)
   Client Characteristic Configuration (0x2902)
- Unknown Characteristic [N W] (00000017-0000-1000-8000-00805f9b34fb)
   Client Characteristic Configuration (0x2902)
- Unknown Characteristic [N WNR] (00000018-0000-1000-8000-00805f9b34fb)
   Client Characteristic Configuration (0x2902)
Unknown Service (00000100-0065-6c62-2e74-6f696d2e696d)
- Unknown Characteristic [WNR] (00000101-0065-6c62-2e74-6f696d2e696d)
- Unknown Characteristic [N] (00000102-0065-6c62-2e74-6f696d2e696d)
   Client Characteristic Configuration (0x2902)
D	20:24:09.909	gatt.setCharacteristicNotification(00002a05-0000-1000-8000-00805f9b34fb, true)
D	20:24:09.910	gatt.setCharacteristicNotification(00002a19-0000-1000-8000-00805f9b34fb, true)
D	20:24:09.911	gatt.setCharacteristicNotification(00000010-0000-1000-8000-00805f9b34fb, true)
D	20:24:09.912	gatt.setCharacteristicNotification(00000019-0000-1000-8000-00805f9b34fb, true)
D	20:24:09.914	gatt.setCharacteristicNotification(00000017-0000-1000-8000-00805f9b34fb, true)
D	20:24:09.916	gatt.setCharacteristicNotification(00000018-0000-1000-8000-00805f9b34fb, true)
D	20:24:09.918	gatt.setCharacteristicNotification(00000102-0065-6c62-2e74-6f696d2e696d, true)
I	20:24:18.912	Notification received from 00000019-0000-1000-8000-00805f9b34fb, value: (0x) 00-00-00-00-01-00
A	20:24:18.912	"(0x) 00-00-00-00-01-00" received
I	20:24:18.961	Notification received from 00000019-0000-1000-8000-00805f9b34fb, value: (0x) 01-00-01-00-00-00
A	20:24:18.961	"(0x) 01-00-01-00-00-00" received

atc1441, 2020-11-27:
Can you please share the chrome console log while activating ?

imbuggy, 2020-11-27:
With MCCGQ02HL.

Connect + Do Activation

[atc1441.github.io-1606507760930.log](https://github.com/atc1441/ATC_MiThermometer/files/5609451/atc1441.github.io-1606507760930.log)

Extract:
acceptAllDevices: trueoptionalServices: Array(4)0: "00010203-0405-0607-0809-0a0b0c0d1912"1: "ebe0ccb0-7a0a-4b0c-8a1a-6ff2997da3a6"2: 651733: 7952length: 4__proto__: Array(0)__proto__: Object
Temp_universal_mi_activate.html:78 {optionalServices: Array(4), acceptAllDevices: true}acceptAllDevices: trueoptionalServices: Array(4)0: "00010203-0405-0607-0809-0a0b0c0d1912"1: "ebe0ccb0-7a0a-4b0c-8a1a-6ff2997da3a6"2: 651733: 7952length: 4__proto__: Array(0)__proto__: Object
Temp_universal_mi_activate.html:386 Found GATT server
Temp_universal_mi_activate.html:309 Services: 0000fe95-0000-1000-8000-00805f9b34fb
Temp_universal_mi_activate.html:386 Status: Detected Mi Thermometer
Temp_universal_mi_activate.html:386 Found Main service
Temp_universal_mi_activate.html:386 Found enc_10 char
Temp_universal_mi_activate.html:386 Found enc_19 char
Temp_universal_mi_activate.html:386 Status: Connected, you can now Do the Activation to either get the Token or flash a new Firmware
Temp_universal_mi_activate.html:386 Send: a2000000
Temp_universal_mi_activate.html:386 Enc_19: 000000000100
Temp_universal_mi_activate.html:386 Send: 00000101
Temp_universal_mi_activate.html:386 Enc_19: 010001000000
Temp_universal_mi_activate.html:386 Send: 00000100
Temp_universal_mi_activate.html:386 Send: 15000000
Temp_universal_mi_activate.html:386

atc1441, 2020-11-27:
Thank you,

That looks like it may has a different activation method. 
I can not suggest into going the route of decrypting it except you have a lot of time, took me about 2 weeks full time to crack it.

You can sniff the stock MiHome acitvation and see what it does so maybe you can compare it to my version but will most likely not be simple

imbuggy, 2020-11-27:
Thank you. I'll see what I can do.

imbuggy, 2020-12-07:
I was able to export a HCIDUMP of the official first 'activation' with this device. I find similarities (I was able to find the public key exchange for example) but it seems to be triggered differently.

Do you see the obvious difference?

ATT Write Request   Packet (17: 01 00)
ATT Write Response Packet
ATT Write Command Packet (16: 24 00 00   00)
ATT Write Command Packet (19: 00 00 00 0B   01 00)
ATT Notification Packet (19: 00 00 01 01)
ATT Write Command Packet (19: 01 00 9F 90   62 BB A1 71 10 00 15 F2 AB F3 8B 79 34 97)
ATT Notification Packet (19: 00 00 01 00)
ATT Notification Packet (19: 00 00 02 0D   A9 63 97 E4 D1 5E 6B 5A DC 47 3F D2 7F 90 1D 7C)
ATT Write Command Packet (19: 00 00 03   00)
ATT Notification Packet (19: 00 00 02 0C   43 75 09 B6 D1 07 5B BF 77 6E 48 B7 5C D3 E5 22 83 1F D8 87 B0 30 C2 0E 79 44   C2 3A 50 8F 71 99)
ATT Write Command Packet (19: 00 00 03   00)
ATT Write Command Packet (19: 00 00 00 0A   01 00)
ATT Notification Packet (19: 00 00 01 01)
ATT Write Command Packet (19: 01 00 06 0D   E3 20 2F 7F 47 5A C9 EA DC EE F1 56 02 23 3A C4 A5 28 BC 14 3C 0F 70 14 74 1F   65 E2 27 81)
ATT Notification Packet (19: 00 00 01 00)
ATT Notification Packet (16: 21 00 00 00)
ATT Read Request Packet (14)
ATT Read Response Packet (31 2E 31 2E 31   5F 30 30 33 35 00 00 00 00 00 00 00 00 00 00)
ATT Write Request Packet (32: 01 00)
ATT Write Response Packet
ATT Write Request Packet (32: 01 00)
ATT Write Response Packet
ATT Write Request Packet (32: 01 00)
ATT Write Response Packet
ATT Write Request Packet (32: 01 00)
ATT Write Response Packet
ATT Write Command Packet (29:   00 00 18 BF E9 7F 9D 82 8C 8D 63)
ATT Write Command Packet (29:   01 00 23 07 E0 7D 8E CE C5 E0 81)
ATT Write Command Packet (29:   02 00 59 89 72 D6 1D DA 0A 21 C6)
ATT Write Command Packet (29:   03 00 1F B3 48 AD 94 72 73 12 2F)
ATT Notification Packet (31:   01 00 B0 E6 D3 9F B9 4A 19 DD DD 42)
ATT Write Request Packet (32: 01 00)
ATT Notification Packet (31:   02 00 E2 DA 35 F7 A1 FF E9 69 60)
ATT Notification Packet (31:   03 00 A9 84 77 A0 57 59 5D 46 87 A1)
ATT Notification Packet (31:   04 00 D1 0C 7D C4 0A ED 80 B7 D6)

[export.xlsx](https://github.com/atc1441/ATC_MiThermometer/files/5655780/export.xlsx)

If not I guess the next step would be to recreate this traffic by attempting to modify your TelinkFlasher.html code, with your permission.

atc1441, 2020-12-08:
Hey just remebered i made this overview while reversing the activation

https://gist.github.com/atc1441/f37b5ad7c1177f7bc91c61010ace9c88

You have my permission to edit the TelinkFlasher of cpurse :), if it works we can maybe make it a real universal actication tool

dnandha, 2021-12-21:
@atc1441 just wanted to leave you a huge thanks! I captured and analyzed the BLE communication of my Mi scooter and after [finding this](https://github.com/atc1441/atc1441.github.io/blob/master/TelinkFlasher.html) I saw that it was exactly the same. I tried the authorization part of your code with on scooter and it got stuck after the key exchange. I found a solution and figured out the other parts like message encryption by reverse engineering. Works!

You can check out my Python package for Mi authentication / communication here: https://github.com/dnandha/miauth
Would appreciate some feedback :)

pauldogg, 2023-11-04:
> Same here with MCCGQ02HL I can connect but it stoped on activation now, please wait.

Same here. Hangs after "activating now"