atc1441/ATC_MiThermometer #109: List of currently compatible devices?

Zixim, 2020-12-02:
Could we create/maintain a list of currently supported devices ?

atc1441, 2020-12-02:
Right now only the LYWSD03MMC is supported.
Or do you talk about WeBluetooth devices ?

Zixim, 2020-12-02:
see, i had no idea about the WeBluetooth devices... that's why a list is useful 😜

atc1441, 2020-12-02:
Sorry WebBluetooth.

Do you talk about supportet thermometers or supported devices that can föash the thermometer ?

Right now only the Thermometer LYWSD03MMC is supportet.

Hope this is clearer now :)

Zixim, 2020-12-02:
I was talking about alternatives to the LYWSD03MMC, which have custom firmware available that we can flash using this method.
And yes, unfortunately for now there's only the LYWSD03MMC.

Thank you very much for this project though. Much appreciated !

atc1441, 2020-12-02:
The next one will most likely be the MHO-C401 but it is very much beta right now https://github.com/atc1441/ATC_MiThermometer/issues/14

It has the same MCU which is mendatory right now to be part of this repo. Other devices with different MCU will need an own repo / project

pvvx, 2021-01-14:
https://pvvx.github.io/MHO_C401/
Over time, I will try to combine it with the alternative firmware MHO-C401...

pvvx, 2021-01-14:
The beta version MHO-C401 is ready. Power consumption is half that of the original...

atc1441, 2021-01-14:
Whoa nice. 

Love your developments :)

ralf-e, 2021-01-14:
Great, now only the LYWSD02 is missing :-)

ralf-e, 2021-01-14:
just tested the beta FW:
Does the C401 behave differently with the beta FW than the LYWSD03 (configured to ATC-Like advertising)?
My 401 here is then recognized by Tasmota as 03 and the battery status display does not work. TAS recognizes the 03 here as ATCxxx.

MHO-C401 (..E5: 34: 5F) with MHO_C401_v12_beta.bin:

LYWSD03 MAC Address A4C138E5345F
LYWSD03 RSSI -53 dBm
LYWSD03 Temperature 24.1 ° C
LYWSD03 Humidity 42.5%
LYWSD03 Dew point 10.5 ° C

ATC MAC Address A4C138FF09DE
ATC RSSI -43 dBm
ATC temperature 24.0 ° C
ATC Humidity 41.0%
ATC dew point 9.9 ° C
ATC Battery 63%

> 22:26:32.156 MI:A4C138E5345F svc[0] UUID (fe95)
22:26:32.156 MI:50305B05755F34E538C1A40A10013D02C00A
22:26:32.157 MI payload unencrypted
22:26:32.157 MI payload length mismatch
22:26:32.161 MI32: slot: 7/16 - ign repeat
22:26:32.168 MI frame 117, last 117
22:26:32.172 MI32: slot: 7/16
22:26:32.173 MI7 payload 0A10013D

> 22:26:38.934 got a newOperation from BLE
22:26:38.943 MI s:7 op:{"BLEOperation":{"opid":"0","stat":"0","state":"IDLE","MAC":"A4C138E5345F","u":"84213767","svc":"ebe0ccb0-7a0a-4b0c-8a...
22:26:38.950 genericSensorReadFn slot 7 res 1
22:26:38.954 readOneSensor reading for slot 7 res 1
22:26:38.966 MI32: scancomplete
22:26:39.965 MQT: tele/tasmota_BLE3_98BFAC/SENSOR = {"Time":"2021-01-14T22:26:39","ATC-df78a2":{"MAC":"a4c138df78a2","Temperature":24.0,"Humidity":41.0,"DewPoint":9.9,"Battery":58,"RSSI":-38},"ATC-0b5b65":{"MAC":"a4c1380b5b65","Temperature":23.9,"Humidity":41.0,"DewPoint":9.8,"Battery":59,"RSSI":-48},"ATC-ead9fb":{"MAC":"a4c138ead9fb","Temperature":23.8,"Humidity":41.0,"DewPoint":9.7,"Battery":62,"RSSI":-41},"LYWSD03-e5345f":{"MAC":"a4c138e5345f","Temperature":24.1,"Humidity":42.5,"DewPoint":10.5,"Battery":null,"RSSI":-43}}
22:26:40.006 readOneSensor - already active reading 7
22:26:40.849 MI op complete context 5050007
22:26:40.849 operation failed -5 for A4C138E5345F
22:26:40.863 MQT: tele/tasmota_BLE3_98BFAC/BLE = {"BLEOperation":{"opid":"17","stat":"-5","state":"FAILNOSERVICE","MAC":"A4C138E5345F","u":"84213767","svc":"ebe0ccb0-7a0a-4b0c-8a1a-6ff2997da3a6","notifychar":"ebe0ccc1-7a0a-4b0c-8a1a-6ff2997da3a6"}}