atc1441/ATC_MiThermometer #111: Feature Request: OTA web page we can run locally

ncgbrown, 2020-12-02:
I have flashed several of these Xiaomi thermometers to put around the house and used your OTA web page for flashing - works great.  Thank you for your work.

The only thing that concerns me - the setting are stored in RAM on the devices - therefore when the battery dies they will have to be reconfigured.  Your web page works great for flashing and settings changes - but is there a way we could get this page in a bin or script that we could run locally?  In a few years when my devices battery dies your web page may not be available anymore.

atc1441, 2020-12-02:
Hey,

You can clone this repo and simply run the telink flasher then https://github.com/atc1441/atc1441.github.io

ncgbrown, 2020-12-02:
Awesome - yes, this works great.  Again - thanks for all the work.

juhjar, 2020-12-03:
Hi,

 tried to run Telink flasher totally offline, wifi and LTE off. What I did...

1. Saved TelinkFlasher.html to computer.
2. Copied TelinkFlasher.html to phone sdcard.
3. Switched off wifi and LTE and run TelinkFlasher.html on Chrome.
4. Made BT connection and waited that temp and humidity were updated.
5. Press "Do Activation" and here it fails.

Device known id: bit.3.129vr6OeZpATC, Mi Token and Mi Bind Key are empty. 
Log says:  Activating now.. and next line is: ReferenceError: sjcl is not defined.

Is there way to make this work completely offline?

Thank You,

-juha

atc1441, 2020-12-03:
You can try right click and save under.

juhjar, 2020-12-03:
> You can try right click and save under.

That's what I did on step 1.

atc1441, 2020-12-03:
Oh ok misunderstoud it. Then you can clone the repo and use it inside the folder. There is one .js file needed in adition to the html file

juhjar, 2020-12-03:
> Then you can clone the repo and use it inside the folder. There is one .js file needed in adition to the html file

I cloned it (download .zip file, there is no file .js

-juha

atc1441, 2020-12-03:
https://github.com/atc1441/ATC_MiThermometer/issues/111

The core.js is needed

juhjar, 2020-12-03:
> #111
> 
> The core.js is needed

1st I copied core.js into phone download folder where is also TelinkFlasher.html. Opened TelinkFlasher.html with Chrome, same result as on posted before.

Then I copied https://github.com/atc1441/atc1441.github.io into phone and opened TelinkFlasher.html in copied folder, same result. 

Device known id: bit.3.129vr6OeZpATC, Mi Token and Mi Bind Key are empty.
Log says: Activating now.. and next line is: ReferenceError: sjcl is not defined.

atc1441, 2020-12-04:
It sounds to me like the security policity on android does not allow to open/include other files from the device which makes sense.
We can now either include the core.js in the telink file or can you try it on a PC?

juhjar, 2020-12-04:
Thanks for reply. I can try laptop after work. It would be more convenient if core.js could be included in Telinkflasher.

Thanks for the help.

-juha

juhjar, 2020-12-05:
Ok, tested on laptop, works fine, so it will be some security issue on Android like you said @atc1441 

Is it difficult/time consuming task to integrate core.js into Telinkflasher?

atc1441, 2020-12-06:
https://github.com/atc1441/atc1441.github.io/blob/master/TelinkFlasher_all_in_one.html

here it is :)

juhjar, 2020-12-06:
> https://github.com/atc1441/atc1441.github.io/blob/master/TelinkFlasher_all_in_one.html
> 
> here it is :)

YES!! tested on phone, wifi and LTE off, works like charm! Big thanks, time for 2nd donation :)

atc1441, 2020-12-06:
Perfect. Glad that i could help :)

closed 2020-12-07 by atc1441