atc1441/ATC_MiThermometer #112: Only transmit OTA temp in celcius

tarheele, 2020-12-02:
Suggestion...

Currently the thermometer transmits the temperature in the unit set for user display with no way to tell in the BLE broadcast which unit is being used. 

I saw that you were making changes currently.  I suggest making this change while hacking the code.

Thanks!

atc1441, 2020-12-02:
Hey. A few weeks back i actually did change the advertising from F when display is set to F to C as someone was iritated by it.

Will leave it like it is for now as someone can always convert it on the receiver side to F if needed 

The changes are not to the firmware right now :)

tarheele, 2020-12-02:
You are awesome!!

It was probably I who was suggesting that before when we talked.

Thanks!

atc1441, 2020-12-02:
Just checked it, here is the link to it https://github.com/atc1441/ATC_MiThermometer/issues/62

closed 2020-12-07 by atc1441