atc1441/ATC_MiThermometer #113: Xiaomi Miija version 2

pablopezunizar, 2020-12-06:
First of all thanks for the great work here! I love this proyect.

This is v2 of the LYWSD03MMC Xiaomi Miija 2, could this be an issue? The rest of the behaviour seems to be normal/accurate. I would like to report that I compared data of a device flashed with the custom firmware against 3 other devices. I have noticed: 

1. an offset in the humidity data (all fine with temperature) of around 6 percent points below the others
2. and a general delay reacting to a change in temperature/humidity. 

If you think this is relevant for the proyect and it worths the time it could take to debug, let me know to provide any other info that could be helpful or test anything else.

[image]

atc1441, 2020-12-06:
Hey. 
Both "Features" are exactly as they are right now. 
The offset is known and you can set your own offset via the webflasher, the sensor data the custom firmware shows is directly from the sensor without filtering

The sensor data is only meassured every minute to save power.

pablopezunizar, 2020-12-06:
Makes sense, thank you @atc1441 !
Closing this.

closed 2020-12-06 by pablopezunizar