atc1441/ATC_MiThermometer #114: Not able to connect to Mi Flora (international)

meierthomas, 2020-12-07:
I have a couple of LYWSD03MMC and using your code was able to flash the alternate firmware. Using an ESP32 (Tasmota 9.x) as a BLE gateway (sensors are too far from the HA machine) I can see the LYWSD03MMC on the ESP32's webpage and in Tasmota.

I also have a few international Mi Fora flower sensors and referencing this guide (https://tasmota.github.io/docs/Bluetooth/) I wanted to use your tool https://atc1441.github.io/TelinkFlasher.html to get the needed "bind_key" as all I can see from the int'l flower sensor on the Tasmota's home page is the RSSI signal strength and MAC address. I read this is due to the fact that the payload is encrypted and I can inject the bind_key and MAC via NRFKey command on the Tasmota command line.

When I connect to the flower sensor I get this message:
NotFoundError: No Services matching UUID <UUID> found in device
Reconnect 1 from 5
NotFoundError: No Services matching UUID <UUID> found in device
Reconnect 2 from 5
NotFoundError: No Services matching UUID <UUID> found in device
Reconnect 3 from 5
NotFoundError: No Services matching UUID <UUID> found in device
Reconnect 4 from 5
NotFoundError: No Services matching UUID <UUID> found in device
Reconnect 5 from 5
Something went wrong, too many reconnect's
Disconnected

Can't I use this tool for this purpose?

atc1441, 2020-12-07:
Hey. 
The WebFlasher only works for the LYWSD03MMC
You can try this version but it will most likely not work either

https://atc1441.github.io/Temp_universal_mi_activate.html

meierthomas, 2020-12-07:
Yeah you're right. It connects saying "Detected Mi Thermometer" following with error "NotFoundError: No Characteristics matching UUID <UUID> found in service with UUID <UUID>...

I didn't want to reflash but simply extract the bind_key... which is then used to configure Tasmota32

Thanks anyway and great tool btw for the LYWSD03MMC

atc1441, 2020-12-07:
Unfortunately then you need to extract the bind key via the mi home app, 

There is a custom apk that can make it easier.

Maybe in future the tool will also support other devices but that part is the hardest of the whole process

Have a great day

TheNitek, 2020-12-07:
AFAIK the MiFlora does not encrypt the payload and therefore does not need a bind key

atc1441, 2020-12-07:
That is even better thank you for the info

michapr, 2020-12-07:
@meierthomas 
You can read all information direct from advertising string.
Only battery information will not be send here.
The sensor does not send all information at once, but each parameter in an other advertising package.

meierthomas, 2020-12-07:
When I look at the MQTT payload no data other than the RSSI value seems to be coming through.

`13:12:24 MQT: tele/tasmota_0924D4/SENSOR = {"Time":"2020-12-07T13:12:24","Flora-6ba71b":{"Temperature":null,"Illuminance":null,"Moisture":null,"Fertility":null,"Battery":null,"RSSI":-53},"TempUnit":"C"}`

Sensor data seems to be coming through when I pair the sensor with the proprietary app so it may be encrypted.

I have raised this on the Tasmota Discord group to see whether I get an answer as this is beyond my skills :)

michapr, 2020-12-07:
it seems to be a question of decoding there...

Using OpenMQTTGateway all is fine:
```
Dec  7 13:35:35 espressif.fritz.box  - O_MQTT_GW-1 - - - Received json : {"id":"C4:7C:8D:6B:66:20","name":"Flower care","rssi":-84,"distance":13.81901,"model":"HHCCJCY01HHCC","fer":1586}
Dec  7 13:36:21 espressif.fritz.box  - O_MQTT_GW-1 - - - Received json : {"id":"C4:7C:8D:6B:66:20","name":"Flower care","rssi":-89,"distance":21.51847,"model":"HHCCJCY01HHCC","tem":20,"tempc":20,"tempf":68}
Dec  7 13:36:42 espressif.fritz.box  - O_MQTT_GW-1 - - - Received json : {"id":"C4:7C:8D:6B:66:20","name":"Flower care","rssi":-84,"distance":13.81901,"model":"HHCCJCY01HHCC","moi":61}
Dec  7 13:37:46 espressif.fritz.box  - O_MQTT_GW-1 - - - Received json : {"id":"C4:7C:8D:6B:66:20","name":"Flower care","rssi":-87,"distance":18.07764,"model":"HHCCJCY01HHCC","lux":128}
```