atc1441/ATC_MiThermometer #120: negative temperatures

hawajrambo, 2020-12-08:
Dear Aaron, 

This is a great hack - congratulations  :-)

Can you please make the thermometer to work on negative temperatures too?

Also displaying the "-" sign somehow, 
and also transmitting the temperature at least in custom advertising mode.

Thank you, 
Gabor

atc1441, 2020-12-08:
Hey. Both should work fine already :)

hawajrambo, 2020-12-08:
wow, good to hear.

if you have a little time, may I ask You, to do a test with the device to put in the fridge in a bag, to see if it actually works in -10 deg Celsius?

Thank you a lot...

atc1441, 2020-12-08:
Yes also that is tested and works without problems

hawajrambo, 2020-12-08:
Hello, 

I reviewed the code, and I see, it can display -9.9 degree indeed.
It could display also any negative degree by not displaying the fraction:    -9.9*C  -10*C
And it could signal the overflow somehow too.
In case of over/underflow the code in show_big_number() just returns, what keeps the last data on the display, and it's not correct.

As my past experience, a few of my thermal sensor damaged after long time (10 years), what causes this overflow.

What is the termal sensor IC in this device?

atc1441, 2020-12-08:
The sensor is an SHTV3 but the SHTC3 is very similar and an datasheet is available for that.

You are right it is kind of a bug that the displayfunction just returns, a better function including lower temps is definitly possible

The degree can be lower thak -9.9 it will be correctly shown in the advertising data

tomtomek9, 2020-12-10:
Hi guys, are you saying it would be fine to use this outdoors then and it should show temps lower than -9.9C via the app?

atc1441, 2020-12-10:
I can not say for the stock firmware and MiHome app but the custom firmware does allow negative temps above -9.9°C
And it was no problem so far to use the Thermometer outside or in the fridge at -20°C

Kyoshiro-san, 2020-12-10:
Hello again !

By the way, when temperatures are negative, the broadcasted value is for example fff8.
Shall I assume ffff is -0°C and fff8 is -0,7°C ?

Thank you :)

atc1441, 2020-12-11:
Hey :)

Thats sounds right yes

Kyoshiro-san, 2020-12-11:
Ok, thank you for confirming :)