atc1441/ATC_MiThermometer #121: Advertising interval not followed ?

Kyoshiro-san, 2020-12-09:
Hi,

I've flashed some XiaoMi LYWSD03MMC devices with the binary firmware available in this repo, and everything seems to work properly, except for advertising interval - unless I missed some key point here.

I'm using the python script on this website to scan for BLE advertisings:
https://hackaday.com/2020/12/08/exploring-custom-firmware-on-xiaomi-thermometers/

It seems to broadcast data every 2s or so:
> Power ON bluetooth device 0
> Bluetooth device 0 is already enabled
> Enable LE scan
> scan params: interval=1280.000ms window=1280.000ms own_bdaddr=public whitelist=no
> socket filter set to ptype=HCI_EVENT_PKT event=LE_META_EVENT
> Listening ...
> 2020-12-09 13:33:47 - Device: A4:C1:38:65:0B:DD Temp: 19.7c Humidity: 42% Batt: 100%
> 2020-12-09 13:33:49 - Device: A4:C1:38:65:0B:DD Temp: 19.7c Humidity: 42% Batt: 100%
> 2020-12-09 13:33:51 - Device: A4:C1:38:65:0B:DD Temp: 19.7c Humidity: 42% Batt: 100%
> 2020-12-09 13:33:55 - Device: A4:C1:38:65:0B:DD Temp: 19.7c Humidity: 42% Batt: 100%
> 2020-12-09 13:33:57 - Device: A4:C1:38:65:0B:DD Temp: 19.6c Humidity: 42% Batt: 100%
> 2020-12-09 13:33:59 - Device: A4:C1:38:65:0B:DD Temp: 19.6c Humidity: 42% Batt: 100%
> 2020-12-09 13:34:01 - Device: A4:C1:38:65:0B:DD Temp: 19.6c Humidity: 42% Batt: 100%
> 2020-12-09 13:34:02 - Device: A4:C1:38:65:0B:DD Temp: 19.6c Humidity: 42% Batt: 100%
> 2020-12-09 13:34:06 - Device: A4:C1:38:65:0B:DD Temp: 19.6c Humidity: 42% Batt: 100%
> 2020-12-09 13:34:08 - Device: A4:C1:38:65:0B:DD Temp: 19.6c Humidity: 42% Batt: 100%
> 2020-12-09 13:34:10 - Device: A4:C1:38:65:0B:DD Temp: 19.6c Humidity: 42% Batt: 100%
> 2020-12-09 13:34:12 - Device: A4:C1:38:65:0B:DD Temp: 19.6c Humidity: 42% Batt: 100%
> 2020-12-09 13:34:14 - Device: A4:C1:38:65:0B:DD Temp: 19.6c Humidity: 42% Batt: 100%

Also, when I connect to the Mi device and the bluetooth icon appears on the device, adversiting seems to stop, and then even after disconnecting, whatever interval I set (10s, 1min, 5min, ...) is not followed.

Am I missing something ?

Thank you for this project ! :-)

atc1441, 2020-12-09:
Hey if you read the getting comfortable part on hackaday it will explain itself.

:)

Kyoshiro-san, 2020-12-09:
Oooh my bad, I should have read that page more thoroughly !

So this setting is changing the frequency the device reads the sensors, and it will affect update interval on both BLE and the display, thus adjusting energy consumption, am I right ?

atc1441, 2020-12-09:
No problem.

The sensor gets read every minute independend from the advertising interval, this already saves a lot of power as the stock firmware meassures every 10 seconds. So setting it to 1 minute is the most ideal values, in the past this feature was more usefull

Kyoshiro-san, 2020-12-09:
I see, thank you for the clarification.
And thank you so much for this work, that's awesome :)

On another note, have you tried using these devices in covered outdoors areas ? I was wondering if they could work or if they'd "pass out" or corrode of potential excess humidity during the night or rainy environment, even under a porch.

atc1441, 2020-12-09:
Yes outdoor should be no problem, and even if the device breaks after 6 month or so its not a big loose.

It can show negative temps without a problem

Kyoshiro-san, 2020-12-09:
Oh that's great !
Thank you so much again :)

closed 2020-12-09 by Kyoshiro-san

deepcoder, 2020-12-09:
Is it correct to equate an increment of the 'frame packet counter' with a new query of the sensor? Do you have any idea why the temperature and humidity sensor reading seem to be a different rate? The graph I published shows this. Thx!

atc1441, 2020-12-09:
Actually a new packet counter is not a new sensor data, this is because some things have changed in the custom firmware, 

in the first versions the sensor was polled every 5-6 seconds but after an edit this was changed to about one minute, so even if the advertising data is at 10 seconds the sensor is only pulled every minute.

it would definitely be a good idea to change the force an update of the sensor at a new packet count

atc1441, 2020-12-09:
We could add a variable that will hold if a new packet count will be and make it an aditional if into this one https://github.com/atc1441/ATC_MiThermometer/blob/master/ATC_Thermometer/app.c#L85

deepcoder, 2020-12-09:
I am now curious what version of firmware I am using.... is there a way to tell this from (1) advertising packet (2) connection to a device or (3) hash or date on my .bin file? If there is not a version stored and queryable on the firmware, this would be a good addition. Even if it was just a simple one byte field that incremented thru 256 values.

atc1441, 2020-12-09:
Yeah good find on that, definitely a needed feature !