atc1441/ATC_MiThermometer #122: PCB pinout / GPIO control

vortigont, 2020-12-10:
Hi! Maybe someone already poked through the P-pads on PCB, is there any pinout available?
Wonder if this tiny could do more than just sending temp/humi? Something like gpio control, door/window sensor, attaching some other i2c sensors maybe? The SDK is quite brief with not much of the examples/libs available to play with.

atc1441, 2020-12-10:
Hey. 

Yes i have a paper at home with most of the pinout. I can post it later if i not forget it

It is definitely possible to add aditional hardware. 

Just as info not much pins are free. 3 or 4 i think

vortigont, 2020-12-10:
Thanks, @atc1441! Would be nice to have pinout available in repo.
Have an idea to use it as water volume counter, all it need is an interrupt-driven event for a single pin open/close.

pvvx, 2020-12-16:
Add to the description that the Sensirion SHTC3 sensor is used.
https://www.sensirion.com/en/environmental-sensors/humidity-sensors/digital-humidity-sensor-shtc3-our-new-standard-for-consumer-electronics/

agmlego, 2020-12-16:
It is unclear how to connect to the board for UART flashing; there seems to be a wire missing from the Mi_SWS_Connection.jpg image. A pinout would be helpful!

pvvx, 2020-12-18:
marking PCB 
[image]

agmlego, 2020-12-18:
Danke so much!

agmlego, 2020-12-18:
Actually, no, now I am more confused. The image for flashing shows a wire connected to P14, which I guess is in the `SWS` position on the mux. Is there no RX needed for flashing?

atc1441, 2020-12-18:
My image was for a non Transceive flasher but now the flasher us update so can also read from the TLSR8251 thanks to Victor

You can use this pinout https://github.com/pvvx/TlsrComSwireWriter

pvvx, 2020-12-18:
BoardPinout
https://github.com/pvvx/ATC_MiThermometer/tree/master/BoardPinout

atc1441, 2020-12-18:
@pvvx Victor to the sensor, inside of the Thermometer is actually an SHTV3 but the SHTC3 is very similar and can be used.

agmlego, 2020-12-18:
Oh! OK, thanks! I got it.

pvvx, 2020-12-18:
> Actually, no, now I am more confused. The image for flashing shows a wire connected to P14, which I guess is in the `SWS` position on the mux. Is there no RX needed for flashing?

There are two options. With and without reading.
Only USB-COM PL-2303HX or CH340 is suitable for reading. Not FTDI!
PL-2303HX is preferred.

pvvx, 2020-12-18:
> @pvvx Victor to the sensor, inside of the Thermometer is actually an SHTV3 but the SHTC3 is very similar and can be used.

Datasheet on SHTV3?
SHTC3 completely matches, including timing and electrical specifications.

pvvx, 2020-12-18:
[image]
[image]
[image]

atc1441, 2020-12-18:
A datasheet is not available for the SHTV3 the sensor works exactly the same so as mentioned the SHTC3 datasheet can be used.

pvvx, 2021-01-04:
[image]
---
[image]
:)

atc1441, 2021-01-05:
Love it!

pvvx, 2021-01-08:
Alarm button (TTP233D-HA6 TonTouchTM', at low power mode typical 2.5uA, maximum 5uA )? :)
http://www.tontek.com.tw/uploads/product/120/TTP233D-HA6_V1.1_EN.pdf

canstb, 2022-05-21:
> ## [image]
> [image] [ ](https://user-images.githubusercontent.com/12629515/103579278-0dc08780-4ee9-11eb-9c24-5e1c6acb58b8.gif) [image] [ ](https://user-images.githubusercontent.com/12629515/103579278-0dc08780-4ee9-11eb-9c24-5e1c6acb58b8.gif) :)

i cant beleive my never ending / never properly working cigar humidifier project could have such simple and cheap solution.
can you please guide me how to get a mi thermometer output active when humidity is below a setpoint like you have posted?