atc1441/ATC_MiThermometer #125: Settings FC01 was sent successfully. But, advertising is not instantly when the temperature changes fast.

rockkoca, 2020-12-11:
Settings FC01 was sent successfully. But, advertising is not instantly when the temperature changes fast.

I used a few MiThermometers with boards only. I put a finger on the sensor. The temperature should have changed from 27+ to 34+ in seconds. But, I did not see instant advertisements. It still reports like every 10s. I did not receive any changes between 27 and 34. I only received 34 after like 20+ seconds. I repeated a few times with no luck to get instant advertising.

I tried two different MiThermometer. It seems the temperature on the LCD also changes very slowly. If I blow warm air on MiThermometer with stock firmware, I can see changes almost instantly on the LCD. But, it seems temperatures are changed very slowly with custom firmware. 

Is this a bug?

Thanks,

rockkoca, 2020-12-11:
I will check the code instead! 
Thanks,

closed 2020-12-11 by rockkoca

atc1441, 2020-12-11:
Hey. This is because in a later update the sensor meassurement was increased to one minute.
This was done to save battery. 

I think it should be lowered again