atc1441/ATC_MiThermometer #126: Connecting sensors to ESP32

miastekpl, 2020-12-11:
Hello, I have 4 sensors LYWSD03MMC. At the moment I have them connected to the Xiaomi Home application through the ZNDMWG02LM gateway. Can I add them to ESP32 and have them in the Xiaomi Home application at the same time?
I want to read Bindkey TelinkFlasher by pressing "Do Activation". I don't know if Bindkey will not change then?
Thank you for your help.

atc1441, 2020-12-11:
Hey. The BindKey does change the moment you click on do activation. 

This is also mentioned in the Telink flasher

KoalaBear84, 2020-12-11:
OK, if it changes then it's not possible to use it at the same time in Xiaomi Home app and ESP32?

Only with the modified Android app? (helping @miastekpl 😇)

atc1441, 2020-12-11:
Yes, both things are correct

The activation with telink flasher cannot be used with any stock app

miastekpl, 2020-12-11:
So, as I understand correctly, it is not possible to connect sensors to Xiaomi Home and ESP32 at the same time?

atc1441, 2020-12-11:
It is possible, 

When the BindKey from the MiHome App is extracted it can be used in the ESP32 and in the MiHome App at once. 

Only my custom Firmware and TelinkFlasher can not be usee with the stock app or Gateway