atc1441/ATC_MiThermometer #127: implausible data get from LYWSD03MMC

ercoleg, 2020-12-11:
Hello,
I'm currently using the following toolchain:
- Two LYWSD03MMC equipped with ATC FW
- One Wemos D1 with Tasmota FW and NRF24 (for advertising information acquisition)
- Transmission of the data via MQTT to Mosquitto on Raspberry Pi
- Python Script to subscribe MQTT topic and  send data to Influx DB
- Grafana to plot data

I usually get good measurements from both sensors on my DB but sometimes, from the farther, I get some implausible information (see image).

I have some questions:
- have you ever seen such behavior (implausible data) in your measurements? 
- is the standard advertising BLE communication protected with some checksum or parity bit?
- is it possible to foresee some protection mechanism in the data sent by the sensor?

Thank you for your help

G

[image]

ercoleg, 2020-12-11:
In this moment, the one that has inplausible data is the one with battery 82%
[image]

satmandu, 2020-12-27:
I'm actually having a similar issue on one of my devices too, which I have outside mywindow (so exposed to ~ 32F weather.)

```
Bluetooth device 0 is already enabled
Enable LE scan
scan params: interval=1280.000ms window=1280.000ms own_bdaddr=public whitelist=no
socket filter set to ptype=HCI_EVENT_PKT event=LE_META_EVENT
Listening ...
2020-12-27 00:19:49 - Device:  A4:C1:AA:BB:CC:DD Temp: 11825.2°F Humidity: 52% Batt: 42%
2020-12-27 00:19:53 - Device:  A4:C1:AA:BB:CC:DD Temp: 11825.2°F Humidity: 52% Batt: 42%
2020-12-27 00:19:57 - Device:  A4:C1:AA:BB:CC:DD Temp: 11825.2°F Humidity: 52% Batt: 42%
```
(The temperature on the device display seems appropriate.)

This is the script I use:
```
#!/usr/bin/env python3
import sys
from datetime import datetime
import bluetooth._bluetooth as bluez
 
from bluetooth_utils import (toggle_device, enable_le_scan,
                             parse_le_advertising_events,
                             disable_le_scan, raw_packet_to_str)
 
# Use 0 for hci0
dev_id = 0
toggle_device(dev_id, True)
 
try:
    sock = bluez.hci_open_dev(dev_id)
except:
    print("Cannot open bluetooth device %i" % dev_id)
    raise
 
# Set filter to "True" to see only one packet per device
enable_le_scan(sock, filter_duplicates=False)
 
try:
    def le_advertise_packet_handler(mac, adv_type, data, rssi):
        data_str = raw_packet_to_str(data)
        # Check for ATC preamble
        if data_str[6:10] == '1a18':
            tempC = int(data_str[22:26], 16) / 10
            tempCst = ("%0.1f°C" % tempC)
            tempF = (9/5)*tempC + 32
            tempFst = ("%0.1f°F" % tempF)
            hum = int(data_str[26:28], 16)
            batt = int(data_str[28:30], 16)
            print("%s - Device: %s Temp: %s Humidity: %s%% Batt: %s%%" % \
                 (datetime.now().strftime("%Y-%m-%d %H:%M:%S"), mac, tempFst, hum, batt))
 
    # Called on new LE packet
    parse_le_advertising_events(sock,
                                handler=le_advertise_packet_handler,
                                debug=False)
# Scan until Ctrl-C
except KeyboardInterrupt:
    disable_le_scan(sock)
```
(This script requires https://github.com/colin-guyon/py-bluetooth-utils/blob/master/bluetooth_utils.py to be in the same folder.)

satmandu, 2020-12-27:
(all of my devices are  LYWSD03MMC equipped with ATC FW)

yeshuawatso, 2021-01-16:
> (all of my devices are LYWSD03MMC equipped with ATC FW)

Your script doesn't conform to the advertisements from the stock ATC FW. You need to bitshift your value in this line:

`tempC = int(data_str[22:26], 16) / 10`

before you divide, maybe something like:

`tempC = (int(data_str[22:26], 16)<<8)/10`

See https://github.com/atc1441/ATC_MiThermometer/blob/a72a5ca20162bc168d27900b6fe15b1847b0f666/ATC_Thermometer/ble.c#L205

ercoleg, 2025-07-05:
Cvo