atc1441/ATC_MiThermometer #129: Wrong bootloader?

rastersoft, 2020-12-12:
I'm checking the SDK documentation (Thank you, google translate!) and found an odd thing: to enable 16K memory retention, there is a single bootloader for 8251, 8253 and 8258 (boot_16k_retn_8251_8253_8258.link); but to enable 32K memory retention there are two: one for the 8251 (boot_32k_retn_8251.link) and another for the 8253 and the 8258 (boot_32k_retn_8253_8258.link).

In your project you use 32K memory retention, but the funny thing is that you use the one for the 8253 and 8258, when, IIRC, the Mi Thermometer uses a 8251...

Am I missing something?

atc1441, 2020-12-12:
Hey. Yes you saw that right.

While developing i checked multible versions and the one used now was the one working.
This is somehow also the version the examples used in the SDK

It would maybe be good to look into that deeper but so far it didnt gave any problems

rastersoft, 2020-12-12:
Well, good to know. I was to try to change it, but if you say that it doesn't work, then better keep it that way. Thanks!

closed 2020-12-12 by rastersoft