atc1441/ATC_MiThermometer #130: Error while flashing Firmware

dittohead90, 2020-12-13:
I'm trying to flash the firmware on the qinping alarm clock but. I used this link which you posted at the thread "Not able to connect to Mi Flora (international)", because with the other one I was unable to connect. 

https://atc1441.github.io/Temp_universal_mi_activate.html

I can do the activition but when I try to flash the firmware I got this error:
"Update error: TypeError: Cannot read property 'writeValue' of null

Any suquestions?
ThankS!

atc1441, 2020-12-13:
Hey. 
The firmware is only for the LYWSD03MMC and no other Thermometer. Thats why it does not work

When the activation worked for you you can use the shown BindKey to use the Thermometer with ESPhome

dittohead90, 2020-12-13:
Okay I see. 
Do you know if I can use the Thermometer then in ioBroker somehow?

atc1441, 2020-12-13:
If ioBroker supports xiaomi thermometer with bindkey it should work but dont know for shure