atc1441/ATC_MiThermometer #132: UUID advertising not showing on esp32 scan

lolren, 2020-12-15:
I'm trying to scan the thermometer with a esp32 for making a simple thermostat. Scanning for the UUID value shows on other devices, but not on the ATC one. This is the ESP32  ble scanner output. 
`Advertised Device: Name: MJ_HT_V1, Address: 58:2d:34:38:be:c2, serviceUUID: 0000180f-0000-1000-8000-00805f9b34fb 
Advertised Device: Name: Mi Smart Band 5, Address: eb:b3:1c:b3:c7:2a, manufacturer data: 570102ffffffffffffffffffffffffffffffff02ebb31cb3c72a, serviceUUID: 0000fee0-0000-1000-8000-00805f9b34fb 
Advertised Device: Name: ATC_9D4132, Address: a4:c1:38:9d:41:32 
Devices found: 3
Scan done! `
Am I missing something ? Kind regards, 

Loren

atc1441, 2020-12-15:
Hey. Have you put it into MiLike advertising via the WebFlasher ?

michapr, 2020-12-15:
> `Device: Name: ATC_9D4132, Address: a4:c1:38:9d:41:32`

Isn't it the device you are looking for ? ;)

lolren, 2020-12-15:
Here is an Idea. If it's not possible to make the UUID show with the ESP32 ( i notice many people have this problem), what if someone modify the firmware to change the name of the device to the UUID ( and we can process the name to get the values out). The mac will remain the same, so it will be easy to filter the contents.Just an idea ....

lolren, 2020-12-15:
> > `Device: Name: ATC_9D4132, Address: a4:c1:38:9d:41:32`
> 
> Isn't it the device you are looking for ? ;)

yes, it is. but, it does not have a corresponding UUID after, like the rest of the devices

lolren, 2020-12-15:
> Hey. Have you put it into MiLike advertising via the WebFlasher ?
Of course. No change. But, I noticed that other people tried to connect it without success with the esp32. 
Would it be possible to advertise the data via name change?

michapr, 2020-12-15:
I use some of these sensors and have no problem with receiving advertising data.
You must not connect to device for this - all will be transferred without connection.

You can use as sample OpenMQTTGateway (or other software)

lolren, 2020-12-15:
> I use some of these sensors and have no problem with receiving advertising data.
> You must not connect to device for this - all will be transferred without connection.
> 
> You can use as sample OpenMQTTGateway (or other software)

I know. and I don't try to connect to the device. Just trying to get the data via UUID advertising. It doesn't work with the esp32.

hallard, 2020-12-16:
I'm using ATC firmware with ESP32 and Micropython (WiPy 3) without any issue, I started with [blescan](https://docs.pycom.io/tutorials/networks/ble/) example and then added decoding advertising data class for the sensors project I needed as stated [here](https://github.com/atc1441/ATC_MiThermometer/#advertising-format-of-the-custom-firmware).

[image]

Sorry decoding code is under NDA, can't share, but that's the idea.

hallard, 2020-12-16:
Just tried with [OpenMQTTGateway](https://github.com/1technophile/OpenMQTTGateway) on ESP32 sending to MQTT Broker works also like a charm (thanks @lolren for sharing this awesome piece of code)

MQTTExplorer Output
[image]

lolren, 2020-12-16:
I see. Guess it`s time to learn "micro"Python then. Thank you for the responses. Regards, Loren

hallard, 2020-12-16:
If you prefer raw C you can check OpenMQTTGateway :-)

lixas, 2020-12-20:
@lolren , This is simple script that shows adv packages on ESP32 micropython terminal. I use it to send known sensors advertisement packages via MQTT messages

NOTE: wait like 30 sec until you see messages in terminal
```
import ubluetooth, ubinascii
from micropython import const

# known BLE list
sensor_mac_list = [
"aa11bb22cc33",
"dd44ee55ff66"
]

def bt_irq(event, data):
    if event == const(6):       # _IRQ_SCAN_DONE
        print("Completed BLE scanning")
    elif event == const(5):     # _IRQ_SCAN_RESULT:
        addr_type, addr, adv_type, rssi, adv_data = data
        # if ubinascii.hexlify(data[1]).decode("utf-8") in sensor_mac_list:       # uncomment to filter known BLE MAC's if needed
        print('type:{} addr:{} rssi:{} data:{}'.format(addr_type, ubinascii.hexlify(addr), rssi, ubinascii.hexlify(adv_data)))
    else:
        print("Terminated")

ble = ubluetooth.BLE()
ble.active(True)
ble.irq(handler=bt_irq)
ble.gap_scan(0, 10000, 10000)
```

lolren, 2020-12-21:
Thank you. I will try it in my esp32 :+1:

DavidStacer, 2020-12-24:
I was able to get the serviceDataUUID and serviceData that contains the advertisement working on a ESP32. 
  https://bitbucket.org/dstacer/workspace/snippets/jBoazB  
My knowledge of C++ is very little so code may not be optimal.

ngxson, 2023-12-08:
> I was able to get the serviceDataUUID and serviceData that contains the advertisement working on a ESP32. https://bitbucket.org/dstacer/workspace/snippets/jBoazB My knowledge of C++ is very little so code may not be optimal.

Thanks for the cue, here is a cleaner version of your script: https://gist.github.com/ngxson/7518f65b9f276dcf6d8c8e6ecd4e8004

The output looks like: `Temperature: 22.10*C - Humidity: 60% - Battery: 59%`