atc1441/ATC_MiThermometer #134: Power consumption when measuring temperature and humidity.

pvvx, 2020-12-16:
Consumption optimization expected?

The total power consumption is reduced when CPU CLK to 16 MHz.
The power consumption for battery measurement is the short measurement cycle. The number of voltage measurements can be increased - these will be small percentages of the total consumption.
The main energy consumption that can be reduced is the temperature and humidity sensor reading cycle.

Original firmware (XiaomiLYWSD03MMC):
[image]

Firmware ATC_Thermometer.bin:
[image]

Test variant (GPIO Wake Up):
[image]

hallard, 2020-12-16:
Awesome, I guess the 3 peaks are while advertising?

pvvx, 2020-12-16:
> Awesome, I guess the 3 peaks are while advertising?
yes.

pvvx, 2020-12-16:
Original firmware (XiaomiLYWSD03MMC):
[image]

michapr, 2020-12-16:
@pvvx Could you provide the code for the sensor.c module (? only here ? ) where you have used the GPIO wakeup?

Thanks!

pvvx, 2020-12-16:
ATC_Thermometr differs in consumption 5..6 times (3.6 mA / 0.6 mA) when measuring temperature. One ATC_Thermometr measurement is equal to 5 measurements Original XiaomiLYWSD03MMC.

> Could you provide the code for the sensor.c module (? only here ? ) where you have used the GPIO wakeup?

If there was a working code, I would have posted it long ago. It has an unresolved problem... 

After sending a command to the sensor for measurement, it is necessary to send the start + sensor address to the bus.
https://www.sensirion.com/fileadmin/user_upload/customers/sensirion/Dokumente/2_Humidity_Sensors/Datasheets/Sensirion_Humidity_Sensors_SHTC3_Datasheet.pdf
SCL to low. Chip goes into deep sleep -> I2C controller reset. On return (wake up), initialization of the I2C controller causes failure of subsequent read data from the sensor.

pvvx, 2020-12-16:
```
#include <stdint.h>
#include "tl_common.h"
#include "drivers.h"
#include "vendor/common/user_config.h"
#include "app_config.h"
#include "drivers/8258/gpio_8258.h"
#include "drivers/8258/pm.h"
#include "stack/ble/ll/ll_pm.h"

#include "i2c.h"
#include "sensor.h"

#define GPIO_WAKEUP_SENS GPIO_PC3 // GPIO_PC2 or GPIO_PC3 ?

int32_t temp;
uint32_t humi;

wakeup_callback_t wakeup_cb; // typedef void (*wakeup_callback_t)(void);

//const uint8_t sens_wakeup[] = {0x35,0x17};
//const uint8_t sens_sleep[] = {0xB0,0x98};
//const uint8_t sens_reset[] = {0x80,0x5D};

void send_sensor(uint16_t cmd) {
	if((reg_clk_en0 & FLD_CLK0_I2C_EN)==0)
			init_i2c();
	reg_i2c_id = 0xE0;
	reg_i2c_adr_dat = cmd;
	reg_i2c_ctrl = FLD_I2C_CMD_START | FLD_I2C_CMD_ID | FLD_I2C_CMD_ADDR | FLD_I2C_CMD_DO | FLD_I2C_CMD_STOP;
	while(reg_i2c_status & FLD_I2C_CMD_BUSY);
}

void init_sensor(){
	send_sensor(0x1735); //	send_i2c(0xE0,(uint8_t *) sens_wakeup, sizeof(sens_wakeup));
	sleep_us(240);
	send_sensor(0x5d80); //send_i2c(0xE0,(uint8_t *)sens_reset, sizeof(sens_reset));
	sleep_us(240);
	send_sensor(0x98b0); //send_i2c(0xE0,(uint8_t *)sens_sleep, sizeof(sens_sleep));
}

_attribute_ram_code_ void read_sensor_start(void) {
	send_sensor(0x1735); //	send_i2c(0xE0,(uint8_t *) sens_wakeup, sizeof(sens_wakeup));
	sleep_us(240);
	reg_i2c_mode &= ~FLD_I2C_HOLD_MASTER;// Enable clock stretching for Sensor
	reg_i2c_id = 0xE0;
	reg_i2c_adr_dat = 0xA27C;
	reg_i2c_ctrl = FLD_I2C_CMD_START | FLD_I2C_CMD_ID | FLD_I2C_CMD_ADDR | FLD_I2C_CMD_DO;
	while(reg_i2c_status & FLD_I2C_CMD_BUSY);
	reg_i2c_id = 0xE0 |  FLD_I2C_WRITE_READ_BIT;
	reg_i2c_ctrl = FLD_I2C_CMD_ID | FLD_I2C_CMD_START;
	while(reg_i2c_status & FLD_I2C_CMD_BUSY);
}

_attribute_ram_code_ void read_sensor_cb(void) {
	int16_t _temp;
	uint16_t _humi;

	wakeup_cb = NULL;
	cpu_set_gpio_wakeup(GPIO_WAKEUP_SENS, Level_High, 0);  // pad high wakeup deepsleep
	init_i2c();
	reg_i2c_id = 0xE0 | FLD_I2C_WRITE_READ_BIT;
//	int i = 10;
//	do {
	reg_i2c_ctrl = FLD_I2C_CMD_ID | FLD_I2C_CMD_START;
	while(reg_i2c_status & FLD_I2C_CMD_BUSY);
//	} while((reg_i2c_status & FLD_I2C_NAK) && i--);

	reg_i2c_ctrl = FLD_I2C_CMD_DI | FLD_I2C_CMD_READ_ID;
	while(reg_i2c_status & FLD_I2C_CMD_BUSY);
	_temp = reg_i2c_di << 8;
	reg_i2c_ctrl = FLD_I2C_CMD_DI | FLD_I2C_CMD_READ_ID;
	while(reg_i2c_status & FLD_I2C_CMD_BUSY);
	_temp |= reg_i2c_di;
	reg_i2c_ctrl = FLD_I2C_CMD_DI | FLD_I2C_CMD_READ_ID;
	temp = 1750*_temp - 450;
	while(reg_i2c_status & FLD_I2C_CMD_BUSY);
	(void)reg_i2c_di;
	reg_i2c_ctrl = FLD_I2C_CMD_DI | FLD_I2C_CMD_READ_ID;
	while(reg_i2c_status & FLD_I2C_CMD_BUSY);
	_humi = reg_i2c_di << 8;
	reg_i2c_ctrl = FLD_I2C_CMD_DI | FLD_I2C_CMD_READ_ID;
	_humi |= reg_i2c_di;
	while(reg_i2c_status & FLD_I2C_CMD_BUSY);
	reg_i2c_ctrl = reg_i2c_ctrl = FLD_I2C_CMD_DI | FLD_I2C_CMD_READ_ID | FLD_I2C_CMD_ACK;
	humi = 100*_humi;
	while(reg_i2c_status & FLD_I2C_CMD_BUSY);
	(void)reg_i2c_di;
	reg_i2c_ctrl = FLD_I2C_CMD_STOP;
	while(reg_i2c_status & FLD_I2C_CMD_BUSY);
//	sleep_us(240);
	send_sensor(0x98b0); //send_i2c(0xE0,(uint8_t *)sens_sleep, sizeof(sens_sleep));
}

_attribute_ram_code_ void read_sensor_deep_sleep(void) {
	read_sensor_start();
	gpio_setup_up_down_resistor(GPIO_PC2, PM_PIN_PULLUP_1M);
	gpio_setup_up_down_resistor(GPIO_PC3, PM_PIN_PULLUP_1M);
	wakeup_cb = read_sensor_cb;
}

_attribute_ram_code_ void read_sensor_go_sleep(void) {
	if(((bls_pm_getSystemWakeupTick() - clock_time())) > 30 * CLOCK_16M_SYS_TIMER_CLK_1MS) {
		cpu_set_gpio_wakeup(GPIO_WAKEUP_SENS, Level_High, 1);  // pad high wakeup deepsleep
		bls_pm_setWakeupSource(PM_WAKEUP_PAD);  //gpio pad wakeup suspend/deepsleep
	} else {
		WaitMs(11);
		read_sensor_cb();
	}
}
```
Other ATC_Thermometr code changes:
```
typedef void (*wakeup_callback_t)(void);

_attribute_ram_code_ void user_init_deepRetn(void){//after sleep this will get executed
	....
	if(pm_is_deepPadWakeup()){
		if (wakeup_cb) {
			wakeup_cb();
			temp += temp_offset;
			humi += humi_offset;
			if((temp-last_temp > temp_alarm_point)||(last_temp-temp > temp_alarm_point)||(humi-last_humi > humi_alarm_point)||(last_humi-humi > humi_alarm_point)){// instant advertise on to much sensor difference
				set_adv_data(temp, humi, battery_level, battery_mv);
			}
			last_temp = temp;
			last_humi = humi;
		}
	}
}

_attribute_ram_code_ int app_suspend_enter(void) {
	if (ota_is_working) {
		bls_pm_setSuspendMask(SUSPEND_DISABLE);
		bls_pm_setManualLatency(0);
		return 0;
	} else {
		if (wakeup_cb) read_sensor_go_sleep();
		bls_pm_setSuspendMask(
				SUSPEND_ADV | DEEPSLEEP_RETENTION_ADV | SUSPEND_CONN
						| DEEPSLEEP_RETENTION_CONN);
		return 1;
	}
}

void main_loop(){	
	blt_sdk_main_loop();
	....
	if (!wakeup_cb)
		read_sensor_deep_sleep();
	....
	app_suspend_enter();
}
```

pvvx, 2020-12-16:
Is timer used in the original firmware?
The current power diagram does not show wake up from GPIO.

pvvx, 2020-12-17:
Turning on and switching the power supply in chip -> pulse on SCL -> bit 15 Temperature lost.
SCL/SDA - PullUp 1M:
[image]

SCL/SDA - PullUp 10K:
[image]

Operation is possible only in the "Clock Stretching mode disabled" for the sensor and deep sleep by timer.

pvvx, 2020-12-17:
Test version with Sensirion SHTC3 humidity sensor working in deep sleep. Timer wake-up.
[image]
Source - https://github.com/pvvx/ATC_MiThermometer
Average consumption 17 μA. Polling of all sensor  every 10..11 sec. Adv.interval 1900 ms.

Connection mode is not finalized. Advertising only.

atc1441, 2020-12-18:
Thank you Victor for looking that deep into it!

Need to buid your Power meter so i can meassure such thinks as well, do you have it documented somewhere? 

Will definitely adapt your changes to make the power consumption go down, or do you want to create a PR?

pvvx, 2020-12-18:
Current consumption in the optimized version:
CR2032 - Start 3.314V 200 mAh.
Adv.Interval ~1.9 sec (3000*0.625 =  1875 ms)
Measured average consumption  (1 h, 3.3V):  17.4 uA
200/0.0174/24/30.5 = **15.7025 months**

Deep-sleep  6 uA
Impulse (Adv): 
Min 3.5 ms only adv TX/RX, average 5.14 mA 
Max 17 ms read sensor, average 1.96 mA
most often impulse  ~4.5 ms

pvvx, 2020-12-18:
> Need to buid your Power meter so i can meassure such thinks as well, do you have it documented somewhere?

For debugging I use:
https://github.com/pvvx/UBIA/tree/master/PowerProfiler
[image]

pvvx, 2020-12-18:
Chrome, USB-COM navigator.serial (experimental-web-platform-features)
https://github.com/pvvx/UBIA/blob/master/WebSerialUSB/tst_wso_adcs_GY-169_power_BLE.gif
Сan write a USB-COM programmer (com-sws) for Chrome and similar explorer...

atc1441, 2020-12-18:
Thats a nice idea !

pvvx, 2020-12-18:
> Will definitely adapt your changes to make the power consumption go down, or do you want to create a PR?

I just shared information. I don’t create final custom versions - no time for maintenance.

pvvx, 2020-12-20:
Power Consumption my new testing (if 2 sec update LCD, measure 10 sec):
Advertising Interval  2 sec - 17..18 uA
Connection (16,16, 99, 800 ->  real interval 2 sec, 3 notify 10 sec) - 15..16 uA 

Advertising use RF TX 3 channel, when connected - 1 channel

michapr, 2020-12-20:
@pvvx Can you compare your results with a 10 Ohm shunt?
For advertising and measure I get some higher values, maybe because of the voltage drop?
(8mA * 100 Ohm = 800mV less voltage on sensor)

pvvx, 2020-12-20:
Maybe. But for the test, relative values are used - comparison with the original. And less needs to be done.
CR2032 also has internal resistance. 
And operating voltage range device 2.0..3.3V. 
Lower voltage will result in higher currents (Internal DC-DC).

pvvx, 2020-12-20:
10 Ohm shunt:
[image]

michapr, 2020-12-20:
>CPU 16 or 24 MHz ?
rf_set_power_level_index(RF_POWER_P3p01dBm)?

I have used your coder from your repo (from yesterday)

For deep sleep: there will be additional current for LCD and sensor (deep sleep) - not much but something.

I want only know the real (true) needed energy for measurement, advertising - this are the parts we could modify over the time... ;)

Thanks!

pvvx, 2020-12-20:
Calculation of the average consumption on the oscilloscope:
[image]
Window 24 ms: Average 1.034 mA, Period (adv.impulse) ~2 sec, Deep-Sleep 6 uA
(1.034mA*24ms+0.006mA*2000ms)/2024ms = 0.0182 mA

michapr, 2020-12-20:
>Window 24 ms: Average 1.034 mA, Period (adv.impulse) ~2 sec, Deep-Sleep 6 uA
(1.034mA * 24ms + 0.006mA * 2000ms) / 2024ms = 0.0182 mA

plus energy of one advertising impulse, right?

pvvx, 2020-12-20:
In the last diagram, this is the maximum duration of an Advertisement with readings of battery voltage and temperature sensor with humidity.

michapr, 2020-12-20:
>It is also necessary to embed a GPIO ("reset" mark on the board) trigger for the assigned temperature / нumidity... + loop recording for a month in flash...
What do you mean with this, sorry, didn't understand ;)

How to set different frequency in code (16MHz)?
(I have checked your TelinkMiFlasher.html (very nice !) but frequency setting is not there, right?)

Thank you for your work and sharing your knowledge!

pvvx, 2020-12-20:
**"Connection" mode.**
[image]
In the "connection" mode, a 10 sec cycle consists of 5 periods (2 sec):
[image]
The transmission is carried out over one channel and the total consumption is less by several percent.
Due to the shorter duration of the active mode.

pvvx, 2020-12-20:
> How to set different frequency in code (16MHz)?

Select only in source code. app_config.h:  #define CLOCK_SYS_CLOCK_HZ  	24000000 / 16000000

---

And don't confuse **CLOCK_16M_SYS_TIMER_CLK_**xx with **CLOCK_SYS_CLOCK_**xx for time settings. System timer always from 16 MHz!

The wrong constant + overflow:
```
if((clock_time()-last_battery_delay) > 5*60000*CLOCK_SYS_CLOCK_1MS){//Read battery delay
```
https://github.com/atc1441/ATC_MiThermometer/blob/master/ATC_Thermometer/app.c#L79

michapr, 2020-12-21:
I have used your repo again ;)
Have you tried the OTA firmware update? I have some problems here (using your flasher page)
At first flashed from original FW to your current release - all fine.
Then have changed the frequency to 16MHz, tried to update via OTA.
```
...
12:14:13: Detected custom Firmware
12:14:15: Start DFU
12:14:21: Update error: NotSupportedError: GATT operation failed for unknown reason.
```
After that device is "dead".
I must use serial update (python) with new compiled FW, then is all again ok. Tried two times, same result.
Any idea? ;)

pvvx, 2020-12-21:
> I have used your repo again ;)

Write there...   Reload 'cstartup_825x.S'

pvvx, 2020-12-21:
[image]

[image]

In different devices and SDKs, the maximum advertising interval is no more than 10 seconds.
For a longer interval, alternate deep-sleep...

michapr, 2020-12-21:
>Write there... Reload 'cstartup_825x.S' (there was a previously changed header)

There are no issues enabled in the repo ;)

ccallawa, 2020-12-21:
I was just looking if there was already an issue filed about the OTA update, but michapr just reported it.  I have the same problem.  Should we create a new issue just for that?

pvvx, 2020-12-21:
The choice of 16 or 24 MHz depends on the number of executable code. If more calculations -> 24 MHz is optimal. Few functions (CPU low load) -> optimal 16 MHz.

> ```
> ...
> 12:14:13: Detected custom Firmware
> 12:14:15: Start DFU
> 12:14:21: Update error: NotSupportedError: GATT operation failed for unknown reason.
> ```
I have a similar problem if change header in firmware.bin. Change cstartup_825x.S.

pvvx, 2020-12-28:
SHTx3 sensor has two measurement commands:

- "Normal" - measurement time is about 11 ms, typical dispersion:

[image]

- “Low Power” - measurement time is about 0.75 ms, typical dispersion: 

[image]

"Low power" measurements are not sufficient to indicate with an accuracy of one tenth of a degree.

atc1441, 2020-12-28:
That looks still very useable with the low power mode

pvvx, 2020-12-29:
Average consumption in ‘connect’ mode, Connect latency: 124 (2.5 sec), Sensor ‘LowPower’ mode: 13.5 uA
[image]

cocktailyogi, 2020-12-30:
Hey Victor!

Nice work. Please keep on going. I am using 8 of this Thermometers in my house and I noticed a quick battery drop over the weeks. I reduction in power consumption is very good. Please let me know, if you need help, maybe in testing.

Kind regards,
Yogi

olivluca, 2020-12-30:
I flashed your version yesterday but it reported the battery at 13% (while with this one is ~90% and dropping fast).

pvvx, 2021-01-01:
> Thats a nice idea !
The USB-COM adapter writes the firmware in explorer. Web version. Connect only TX-SWS and GND wires.
[image]
https://github.com/pvvx/ATC_MiThermometer/blob/master/USBCOMFlashTx.html

hallard, 2022-10-16:
Sorry guys for this topic revival, always impressed by the deep job made and published here, respect !!!

We're working on adding CGG1 PVVX and ATC1441 frames into OpenMQTTGateway  [here](https://github.com/1technophile/OpenMQTTGateway/issues/1286) and looks like it is working fine.

Has somebody tested the implication of peak current when changing default `RF TX Power`?
Mine is set to `VANT+0.04 dbm` (default) but I was wondering what can be battery draining more selecting `VANT+0.58 dbm` for example 
I see there @pvvx has `VANT+3.01 dbm`, is it the default settings with stock firmware?

Thanks

pvvx, 2022-10-16:
My [TelinkMiFlasher](https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html) has a TX RF switch for high power, up to +10 dB. To do this, you need to change the [value](https://github.com/pvvx/pvvx.github.io/blob/master/ATC_MiThermometer/TelinkMiFlasher.html#L30):
_const MAX_RF_TX_Power = true;_
To work with more than + 3Db, the CR2032 battery cannot supply the required current.
At more than +3dB, the Soc switches power directly to the battery. Below +3dB, the internal DC-DC is used.
In the end, the return drops much lower than with negative dB.

hallard, 2022-10-16:
Thanks, interesting, since CGG1 has CR2430 battery it could be better. And now I think I got what is usage of 4.7uH self on the board for DC/DC?
So from your experience, does it worth increasing `VANT` to +1 or +2 db (to increase range) instead of 0.04 default in term of battery life? if it divide battery life per 2, not solution on my side, I'm trying to get a good range and maximise battery life knowing one impact the oher.

pvvx, 2022-10-16:
Increasing RF power affects non-linearly. It is difficult to give an exact ratio, because the total consumption depends not only on the power of the transmitter.

hallard, 2022-10-16:
Yeah I read all this post, also sensor :)
just tried to get an idea, but real life means Ii need to change battery and parameters on same time on my 3 devices to test, that's it :-)

pvvx, 2022-10-16:
> Thanks, interesting, since CGG1 has CR2430 battery it could be better. And now I think I got what is usage of 4.7uH self on the board for DC/DC? 

This is a step-down regulator for the internal power supply of the microcircuit.

The CR2430 is also missing +10 dB. There, the surge current can reach more than 35 mA. This will heavily impulsively load the battery and it will be difficult to connect to the device.