atc1441/ATC_MiThermometer #135: qingping CGDK2 Temp & RH Monitor Lite

michapr, 2020-12-17:
Hi,

I have got two such devices, compatible to Mi Home app.
https://www.aliexpress.com/i/1005001868991135.html
Sensor have LCD display, is not advertising temp/hum, I think every 20minutes about sending an encrypted advertising package.

Sensor is using TLSR8253 and for LCD a BU9792AFUV
https://datasheet4u.com/datasheet-parts/BU9792FUV-datasheet.php?id=705112

What chip is using LYWSD03MMC  for encoding? 
Maybe it is compatible (I think that not...) or driver maybe adapted...

Not sure in this time, interesting or not ... ;)

atc1441, 2020-12-17:
Sounds like a good candide. The LYWSD03MMC is using an TLSR8251 so should be mostly compatible

michapr, 2020-12-17:
Problem can be the LCD driver... ;)

atc1441, 2020-12-17:
That can be reverse engeneered by listening to the gpio transmitting so i dont see a problem there.

And sorry for misunderstanding it first. 
The LCD driver on the LYWSD03MMC is unknown as it has not findable markings on the chip. I got it working via reversing as well, so testing every segment one by one etc

michapr, 2020-12-18:
Just have seen that after "activating" the device in Qingping app (add it once to the device list) the device is advertising the values not decrypted more. - multiple times per minute (every 5 seconds ?).
After closing app (shutdown the mobile phone - it is a test phone I'm using for this)  and removing battery of device, waiting and put it in again the device is still reporting all.
That's interesting for me - and maybe for some others too.

Advertising data now: 8810089211342d580104e2007901020162
```
8810 089211342d58 01 04 e2 00  79 01 0201 62
xxxx   MAC-addr.  xx xx  temp   hum  fw?  batt
```
```
temp = 0*256+e2H = 226 -> 22.6 degree
hum  = 01*256 + 79H = 377 -> 37.7%
batt = 62H = 98%
about firmware relase not sure (current: 1.02) 
```
So in this time seems there are no needs to change the firmware here ;)

atc1441, 2020-12-18:
Thank you for the infos

pvvx, 2020-12-18:
[image]
Is the mi_release_version_4_1_1 SDK version not for this device?
https://yadi.sk/d/g5fV7WD1EaUdRQ
Earlier 1/14/20 on Telink website

pvvx, 2020-12-18:
> The LCD driver on the LYWSD03MMC is unknown as it has not findable markings on the chip.
IST3055NA0 - Datasheet not found
https://github.com/pvvx/ATC_MiThermometer/blob/master/BoardPinout/PcbSide1.jpg
Duplication of information: https://github.com/atc1441/ATC_MiThermometer/issues/122#issuecomment-747856364

michapr, 2020-12-18:
@pvvx >Is the mi_release_version_4_1_1 SDK version not for this device?

Not sure, cannot find the LCD driver here - but maybe my fault.

pvvx, 2020-12-18:
There are no drivers, no "libmijia_std_authen_eclipse_telink_1_0_1.a" (provided by xiaomi). But there is cryptography and xiaomi profiles ...

tjheikki, 2020-12-30:
@michapr  > Just have seen that after "activating" the device in Qingping app (add it once to the device list) the device is advertising the values not decrypted more. - multiple times per minute (every 5 seconds ?).
> After closing app (shutdown the mobile phone - it is a test phone I'm using for this) and removing battery of device, waiting and put it in again the device is still reporting all.
> That's interesting for me - and maybe for some others too.

Zillion thanks, Michael! Got success with two CGDK2'es I thought were destined to be (lousy) fishnet weights. First tried with Xiaomi app, no luck, but after pairing with Qingping app and power cycling they have been sending unencrypted beacon messages for hours now. Firmware 1.02 apparently.
The message byte order seems to be the same than the non-lite CGG1-Qingping/Cleargrass devices (nRF52832-based) send, those just didn't require this pairing step to send usable data.
Messages samples from two CGDK2's and one CGG1:
CGDK2-1: 58:2d:34:11:94:b4 RSSI:-76 payl: 2 1 6 14 16 CD FD 88 10 B4 94 11 34 2D 58 1 4 **D8 0 BB 1 2 1 59** 18 9 51 69 6E 67 70 69 6E 67 20 54 65 6D 70 20 26 20 52 48 20 4C 69 74 65  temp:21.6 rh:44.3
CGDK2-2: 58:2d:34:11:92:7b RSSI:-57 payl: 2 1 6 14 16 CD FD 88 10 7B 92 11 34 2D 58 1 4 **D4 0 CB 1 2 1 5C** 18 9 51 69 6E 67 70 69 6E 67 20 54 65 6D 70 20 26 20 52 48 20 4C 69 74 65  temp:21.2 rh:45.9
CGG1:    58:2d:34:11:08:84 RSSI:-70 payl: 2 1 6 14 16 CD FD 8 7 84 8 11 34 2D 58 1 4 **D8 0 C3 1 2 1 46** 15 9 43 6C 65 61 72 47 72 61 73 73 20 54 65 6D 70 20 26 20 52 48  temp:21.6 rh:45.1

The byte after Qingping's manufacturer ID "CD FD" (0xFDCD) is always "8" in CGG1 (I have several) but varies in CGDK2.

I am an embedded developer but no experience with TLSR82*, thanks to this interesting project I'll get some. I can look at this CGDK2 if needed. but now looks maybe not. The E-ink CGG1 would be an interesting platform, reverse engineering that one is on my list.

TheGroundZero, 2021-02-04:
I got myself some Qingping CGG1 in the hopes of getting them flashed and integrated in Home Assistant.

It appears these cannot be flashed (yet) with the available firmware?

Connecting with the device from the Web Flasher
```
Log:
23:11:21: Disconnected.
23:11:21: Searching for devices
23:11:29: Connecting to: Qingping Temp & RH M
23:11:32: Connected
```

Trying to Activate
```
23:11:37: Not connected
```

atc1441, 2021-02-05:
Hey. 
Yes as in the other issue mentioned they are in itself very different.

It is possible to create a custom firmware for them but someone has to do it :)

truethe1mc, 2021-02-11:
> > Just have seen that after "activating" the device in Qingping app (add it once to the device list) the device is advertising the values not decrypted more. - multiple times per minute (every 5 seconds ?).
> > After closing app (shutdown the mobile phone - it is a test phone I'm using for this) and removing battery of device, waiting and put it in again the device is still reporting all.
> > That's interesting for me - and maybe for some others too.
> 
> Zillion thanks, @michapr! Got success with two CGDK2'es I thought were destined to be (lousy) fishnet weights. First tried with Xiaomi app, no luck, but after pairing with Qingping app and power cycling they have been sending unencrypted beacon messages for hours now. Firmware 1.02 apparently.
> The message byte order seems to be the same than the non-lite CGG1-Qingping/Cleargrass devices (nRF52832-based) send, those just didn't require this pairing step to send usable data.
> Messages samples from two CGDK2's and one CGG1:
> CGDK2-1: 58:2d:34:11:94:b4 RSSI:-76 payl: 2 1 6 14 16 CD FD 88 10 B4 94 11 34 2D 58 1 4 **D8 0 BB 1 2 1 59** 18 9 51 69 6E 67 70 69 6E 67 20 54 65 6D 70 20 26 20 52 48 20 4C 69 74 65 temp:21.6 rh:44.3
> CGDK2-2: 58:2d:34:11:92:7b RSSI:-57 payl: 2 1 6 14 16 CD FD 88 10 7B 92 11 34 2D 58 1 4 **D4 0 CB 1 2 1 5C** 18 9 51 69 6E 67 70 69 6E 67 20 54 65 6D 70 20 26 20 52 48 20 4C 69 74 65 temp:21.2 rh:45.9
> CGG1: 58:2d:34:11:08:84 RSSI:-70 payl: 2 1 6 14 16 CD FD 8 7 84 8 11 34 2D 58 1 4 **D8 0 C3 1 2 1 46** 15 9 43 6C 65 61 72 47 72 61 73 73 20 54 65 6D 70 20 26 20 52 48 temp:21.6 rh:45.1
> 
> The byte after Qingping's manufacturer ID "CD FD" (0xFDCD) is always "8" in CGG1 (I have several) but varies in CGDK2.
> 
> I am an embedded developer but no experience with TLSR82*, thanks to this interesting project I'll get some. I can look at this CGDK2 if needed. but now looks maybe not. The E-ink CGG1 would be an interesting platform, reverse engineering that one is on my list.

@tjheikki could you say how you read advertising from CGDK2?
for example while connecting with gatttool to LYWSD03MMC, it does send notifications with temp/hum very frequently, while connecting to CGDK2 does not give any notifications, i only could retreive the battery level there ( similar to LYWSD03MMC)

tjheikki, 2021-02-11:
@truethe1mc Did you pair with the "Qingping app" and power cycle? Worked for me with two devices and they have been constantly sending temp & humi since, for about 6 weeks now. Don't remember how often they send, more frequent than every  five minutes, which is my gateway's polling frequency.

Guess they have two different modes; one for Xiaomi app, perhaps only working paired, and another for the Qingping, where it sends the data as non-encrypted advertising packets. My 8-ish CGG1s did the latter out-of-the-box, but those apparenlty have differently working firmware versions.

truethe1mc, 2021-02-11:
@tjheikki thanks for your fast reply.
By the power cycle you mean put the battery off, wait and put it back? Yes, i did try that, but didnt notice any difference and the tool i use to check these advertising notifications in Linux ( **gatttool**) didn't show any notifications.

What i've noticed is when CGD2 is paired with MiHome , 16b Service Data which **sudo blescan** shows very similar with the same string for LYWSD03MMC (something like 95fe****5b********75**c1a**8, 28 chars long), which leads me to an idea that paired with MiHome, CGD2 should work somehow similar to LYWSD03MMC
While pairing with Qingping app ( Chineese region) gives another 16b Service Data pattern, looks like then the sensor acts somehow differently( like cd****10******342******4f200******0156, 38 chars longs)

spiri439, 2021-06-20:
Any updates on this, I have succesfully flashed the custom firmware and see data from the cgdk2, but nothing is displayed.

mvdklip, 2021-12-13:
So am I right in thinking that getting the ATC firmware to run on these CGDK2 devices "only" requires hardware detection and a different LCD driver?

@spiri439 How did you flash the ATC firmware on this device?

mvdklip, 2021-12-13:
> https://esphome.io/changelog/v1.19.0.html
> 
> [esphome/esphome#1451](https://github.com/esphome/esphome/pull/1451)

This is about being able to parse the CGDK2 output using a device running the esphome firmware. It is not about being able to run custom firmware on this device.

unaiur, 2022-01-21:
I also bought some of these thermometer by mistake (I did not realise about the "Lite" suffix). The LCD driver seems quite easy: the data sheet is almost the same than BU9797FUV that is available in English (it has some extra options for some commands but it is almost the same).

The hard part is knowing how is everything connected: to which PIN is connected the KEY, the sensor I2C and the display I2C (maybe they share a single I2C bus) and if there is any other PIN that needs to be configured HIGH for the display to work.

@spiri439, you said that you flashed a firmware and you get the sensor working... which firmware did you flash? Because each device uses different ports for sensor I2C.

I will write some code to detect if the display is in the same I2C bus, but I need to know which firmware did you flashed.

pvvx, 2022-01-22:
https://aliexpress.com/item/1005002634860799.html
I bought a few pieces to test and possibly support custom firmware. The order will arrive in a month...

unaiur, 2022-01-24:
I will extract the firmware but, is there any tool to disassemble it and learn how the GPIOs are configured?

pvvx, 2022-01-25:
Good photo of the board from both sides + datasheet: 'shows the pin assignment' :
http://wiki.telink-semi.cn/wiki/chip-series/TLSR825x-Series/

unaiur, 2022-01-26:
I will try to open it tonight…

unaiur, 2022-01-26:
I've found these photos from the fcc registry: https://fccid.io/2AQ3F-CGDK2/Internal-Photos/Internal-Photos-4716361.pdf

unaiur, 2022-01-26:
And these are the photos of my device:

https://share.icloud.com/photos/09asHIA9n8xlDzkfAkIc8A6Iw

https://share.icloud.com/photos/0862Rxkd4kzv-0xz32-SoHDLA

unaiur, 2022-01-26:
I make a last photo with maximum exposure and it is easier to see the pcb lines:

https://share.icloud.com/photos/06a-QTMkplOKDEDuzNtu6jVhw

pvvx, 2022-01-27:
Program only uses I2C pins and 'Key'.

<html xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:dt="uuid:C2F41010-65B3-11d1-A29F-00AA00C14882"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=OneNote.File>
<meta name=Generator content="Microsoft OneNote 14">
</head>

<body lang=ru>

<div style='direction:ltr'>

SoC pin | Name | Mark
-- | -- | --
1 | SWM | TP3?
2 | UART_TX | TP1?
3 | UART_RX | TP17?
20 | I2C_SDA |  
21 | I2C_SCK |  
24 | PC4/Key | TP11

</div>

</body>

</html>

unaiur, 2022-01-31:
Umm, @spiri439 said that ATC_MiThermometer worked for him but it cannot be possible: this firmware (atc1441) only supports LYWSD03MMC, which uses different SoC pin for I2C. Maybe he commented in the wrong project and used CGG1 firmware from @pvvx project... Or maybe he just checked that BLE was working... who knows...

unaiur, 2022-01-31:
Seems that this device uses a subset of CGG1 pins. It also uses a version of the chip with less pins. So CGG1 firmware should work perfectly except for the LCD display. I think it will be better to start with Victor's firmware.
I will try to add a generic BLE method to send arbitrary commands to the display and experiment...

unaiur, 2022-02-01:
Ohh, Victor's firmware already has a method to write to the lcd memory. Is there any instructions about how to use it, @pvvx?

pvvx, 2022-02-01:
> Seems that this device uses a subset of CGG1 pins. So CGG1 firmware should work perfectly except for the LCD display. I think it will be better to start with Victor's firmware.
I will try to add a generic BLE method to send arbitrary commands to the display and experiment...

It's easier to connect a logger and record how the LCD controller is configured there and what commands work out the image change.
You can endlessly select controller settings without knowing exactly what the LCD matrix itself is.

pvvx, 2022-02-09:
CGDK2 exists in 3 versions:
1. CGDK2-2 with LCD - no BLE (Qingping CGDK2 Temp & RH Monitor Lite LE)
2. CGDK2 with LCD (Qingping CGDK2 Temp & RH Monitor Lite Mi)
3. CGDK2 with E-ink (?).

unaiur, 2022-02-10:
I see that @pvvx already committed the support for this device to his repository!

unaiur, 2022-02-16:
> Average power consumption - over 48.9 uA !
> *Maximum battery life 8 months!*
> (290mAh/0.0489mA/24h/30.5m = 8.1 months)

Hi @pvvx, had you seen my two pull requests in your repo?

pvvx, 2022-02-16:
Yes, saw. and commented.
Support for CGDK2 is already built into the system in another implementation.

ltegzes, 2022-03-05:
Hi @pvvx,  I try to flash CGDk2 using UART PL2303HX, because I did an error in the course of OTA flashing. I am unsuccessful. There is no proper communication between the TLSR8253 and the UART. I try to use your TlsrPgm.py. 

jive@MacMini Flash % python3 TlsrPgm.py -s -p /dev/tty.usbserial-2140 i
=======================================================
TLSR82xx TlsrPgm version 27.04.21
-------------------------------------------------------
Open /dev/tty.usbserial-2140, 230400 bit/s... ok
Error[102] Read PGM Version and Config!

I use the following connection:

CGDk2 ---- UART

 SWS ----->  RX
 SWS <--1.8 kohm resistor <--- TX
 GDN ------> V -
   V+ ------> 3.3 V

Is there anything I did incorrectly or there is anything extra I should do to be able to flash your program into CGDk2 ?

ltegzes, 2022-03-05:
Corrected the connection:

Hi @pvvx, I try to flash CGDk2 using UART PL2303HX, because I did an error in the course of OTA flashing. I am unsuccessful. There is no proper communication between the TLSR8253 and the UART. I try to use your TlsrPgm.py.

jive@MacMini Flash % python3 TlsrPgm.py -s -p /dev/tty.usbserial-2140 i

TLSR82xx TlsrPgm version 27.04.21

Open /dev/tty.usbserial-2140, 230400 bit/s... ok
Error[102] Read PGM Version and Config!

I use the following connection:

CGDk2 ---- UART

SWS -----> RX
SWS <--1.8 kohm resistor <--- TX
V- ------> GDN
V+ ------> 3.3 V

pvvx, 2022-03-05:
Option 1: To work with USB-COM is used **TlsrComProg**.py from https://github.com/pvvx/TlsrComProg825x
```
python3 TlsrComProg.py -p /dev/tty.usbserial-2140 -t5000 wf 0 CGDK2_v37a.bin
```

---

Option 2: In Windows - [The USB-COM adapter writes the firmware in explorer. Web version.](https://github.com/pvvx/ATC_MiThermometer#the-usb-com-adapter-writes-the-firmware-in-explorer-web-version) - used [USBCOMFlashTx.html](https://pvvx.github.io/ATC_MiThermometer/USBCOMFlashTx.html)
Connect only TX-SWS and GND wires.

---

Option 3: [USB to UART](https://github.com/atc1441/ATC_MiThermometer#usb-to-uart)

ltegzes, 2022-03-05:
Hi @pvvx, thank you, I'll try.

arohl, 2023-01-03:
I am trying to add this device to my home assistant setup. It is found by Telink Flasher v4.7 and I can connect to it. However, click "Do Activation" doesn't appear to do anything - it just says "Activating now, please wait..."

I see that there is custom firmware available, so what am I doing wrong?

I have done this successfully on a couple of Xiaomi LYWSD03MMC so am familiar with the process.

Many thanks

  Andrew