atc1441/ATC_MiThermometer #136: Optional broadcast format of BLE RUUVI tags

PA7T, 2020-12-19:
Dear @atc1441,

impressive work! I just flashed one of mine and everything works like a charm...
I am wondering, if you are aware of RUUVI tags, they are also BLE H+T sensors, which send their data via the braodcast.
These sensors are working in my home since years without change of battery.

It would be great to make an option, so the MiThermometer broadcasts in the same format as the RUUVI tags, to immediately establish compatibility and use all the readout software/plugins for the RUUVI tags.

The format is described under:

https://github.com/ruuvi/ruuvi-sensor-protocols/blob/master/dataformat_05.md

I could imagine the implemention is easy for you, being deep in the matter right now!

Cheers,

PA7T / DL2ZZ

gingerlime, 2021-02-16:
+1 I'm also using some ruuvi tags and would be great if it was possible to broadcast in the same format.

I'm still trying to figure out how to listen to the advertised data and push it to mqtt but couldn't find anything yet... (I know it's a very noob question, but I'm definitely new to all this)

PA7T, 2021-02-17:
Hey gingerlime,

You could make a BLE to MQTT gateway with:

[esphome xiamo ble](https://esphome.io/components/sensor/xiaomi_ble.html?highlight=lywsd03mmc)

and 

[esphome mqtt](https://esphome.io/components/mqtt.html?highlight=mqtt)

on a esp32 board.

gingerlime, 2021-02-17:
Thanks @PA7T. I already have a raspberry pi that already runs a [mqtt gateway](https://github.com/zewelor/bt-mqtt-gateway) for ruuvi, so was hoping to use the same one for these sensors as well... It's funny, but I didn't think to check the gateway I'm using, and it already supports this sensor 🤦 🍾