atc1441/ATC_MiThermometer #138: Advertising interval sticks to 60s (need 5min)

attley86, 2020-12-21:
Hello,

I noticed that this issue was already placed: #33 (Issue closed) and I think there is no way to get the 5min or 10min interval yet?

I would appreciate if someone could compile a binary for me that has: MiLike advertising, no smiley, battery level disabled and interval of 150sec (or whatever is the maximum closest to the 5min that can be configured).

Cheers,
Olli

attley86, 2020-12-21:
In fact when I tried to configure 5min interval from the web page, the interval is even shorter:

```
[08:07:35][D][xiaomi_ble:300]: Got Xiaomi LYWSD03MMC (A4:C1:38:43:94:04):
[08:07:35][D][xiaomi_ble:303]:   Temperature: 15.1°C
[08:07:35][D][xiaomi_ble:306]:   Humidity: 44.0%
[08:07:58][D][xiaomi_ble:300]: Got Xiaomi LYWSD03MMC (A4:C1:38:43:94:04):
[08:07:58][D][xiaomi_ble:303]:   Temperature: 15.1°C
[08:07:58][D][xiaomi_ble:306]:   Humidity: 44.0%

```

atc1441, 2020-12-21:
Hey  
The real BLE interval will allways be about 3-4 seconds, only the message will be updated when the set interval is reached.

Because of the buffer overflow on higher time it will fail as you have seen, but it will still send every 3-4 seconds a new advertising

attley86, 2020-12-21:
Ok. Is there a way to make battery life longer in any other way? I cannot achieve longer message update period with the web page parameters.

atc1441, 2020-12-21:
As mentioned the battery life will not get longer when increasing the advertising time. There is développement going on on to increase the battery life

closed 2020-12-21 by attley86