atc1441/ATC_MiThermometer #140: measure and bake in offsets

ke-ss, 2020-12-23:
This is nice work, thank you :)

I have 4 new sensors. before flashing all read the same values.
I flashed mine in a stable temp and humidity environment.
After flash the temp and % RH  were different, indicating devices have been calibrated before baked with flash.

Suggestion:
1. advise maker to keepin very stable environment just before and just after flashing
2. Before flash measure temp and %RH      and       measure offset being used.
3. Before or after flash, apply offset so that device is corrected.
It may be that a comparison between before flash and after flash temps has to be made to guage the offset, and then another flash, incorporating the correction made.   

Notiuced that phone only displays currently temp and %RH when original firmware is used before flash, but not after. Would be nice oif can work after flash with custom firmware.

Nice code :) Thank you

atc1441, 2020-12-23:
Hey. The offset is known, is always about 4-5%
You can set a custom offset in the webflasher for each device. Which will last till battery is pulled out as you already know.

I have the persistence storage high in my todo list but beeing very ocupied by another project right now.

ke-ss, 2020-12-23:
Hi,
I did notice the % RH was 5% down with the last of the 4 I flashed with you rom.
Since it is known each sensor needs +5 %RH for correct calibration, why not just bake it into the rom ?
Feels like quite an easy fix.

Thaks again, great work :)

atc1441, 2020-12-23:
The main reason why its not baked in is its not known what the real value is, i am using the one directly comming from the sensor and schouldnt that be the correct one ? 
Xiaomi build something into the firmware which does the offset but it is not clear why.

When someone has a validated device to test the real % we could follow that one

giannoug, 2020-12-26:
I just got a dehumidifier and decided to cross check measurements. It seems that the 5% isn't needed, at least at about 60% RH. My dehumidifier shows the exact same value as the stock configuration of this firmware, while any other device I have with +5% are, well, 5% up 🤔