atc1441/ATC_MiThermometer #141: How to find MAC

pristis, 2020-12-23:
Hello!

The firmware works perfect on LYWSD03MMC! Thank you!

I am using home automation and I want to add more than one temp sensors, so I need MAC addresses of them to assign for every room. So, please, do you know how to find MAC addresses of the sensors?

Best regards,
Martin

KoalaBear84, 2020-12-23:
You can even look at the Telink Flasher page, stand close to the one you want to know, and see which one has the best signal.

Or take out the battery and you see 3 times 2 characters displayed which make up the last 6 characters of the MAC address.

https://github.com/atc1441/ATC_MiThermometer#getting-the-mac-of-your-thermometer

mhertel, 2020-12-23:
And the mac address starts with A4:C1:38

pristis, 2020-12-23:
That was very useful! Thank you very much, guys!

lanrat, 2021-01-05:
Would it possible to log or display the device's full mac in the web flasher?

atc1441, 2021-01-05:
On stock firmware, Unfortunately, this is not easy as it is a security risk for some. 

And today most devices only have fake MACs so WebBluetooth will most likely never get that feature

On the custom firmware the  name could be used for that

thirteeneast, 2022-03-04:
Helped me too, thanks.