atc1441/ATC_MiThermometer #142: NotSupportedError: GATT Error Unknown

devzwf, 2020-12-24:
Hello 
I flash with your  custom firmware :ATC_thermometer.bin
All went well , device restart and i can See the last 3 Digit of the MAC
But i cant connect to it ater that with the App 
i see them and when i hit pair on ATC_XXXXXX
I get in the log 
"NotSupportedError: GATT Error Unknown"

On my ESPHome i can see the device but of course do not report anything

[23:40:12][D][esp32_ble_tracker:544]: Found device XX:XX:XX:XX:XX:XX RSSI=-57
[23:40:12][D][esp32_ble_tracker:565]:   Address Type: PUBLIC
[23:40:12][D][esp32_ble_tracker:567]:   Name: 'ATC_XXXXXX'

I change the mac here  for XXXXXXXXX

It is not my first device flash i did few of them last month with no issue

devzwf, 2020-12-24:
i am using the bin form Nov14

atc1441, 2020-12-24:
Please try to reflash the file once

devzwf, 2020-12-24:
How,
i cant connect anymore.
But the device seems to work

devzwf, 2020-12-24:
I tried on a second one and same result. should i try with the bin from 20 Nov (seems the bin file is new)

atc1441, 2020-12-24:
Then please try to restart the pc/phone that you use the webFlasher with and the device itself as well and try again.

Every file should work without a problem

devzwf, 2020-12-25:
Hello , 
Some Update,  i reflash one of the device using serial, same file and all working fine....
I will shutdown the phone and restart it and retry on the second device,  i will let you know what was the result .

Thanks for all the great work

dwhacks, 2021-01-07:
I had this same issue just now. After restarting the phone I used to flash it all worked perfect and was happy. What a great project, thank you!!!

saiajin, 2021-05-14:
Hi I am having similar issue when trying to flash:

1:09:28 AM: Load firmware file 'ATC_Thermometer31.bin'...
1:09:28 AM: File size: 61956 bytes
1:09:28 AM: Count: 3873
1:09:46 AM: Start DFU
1:10:48 AM: Update error: NotSupportedError: GATT operation failed for unknown reason.

elalemanyo, 2021-09-27:
I am getting the same error, is there a solution?

Thanks

devzwf, 2021-09-27:
restart the device  that you use the webFlasher with and the device itself as well and try again.
also as Aaron himself note in his readme :
`Victor @pvvx did some very nice refinings of this custom firmare so i really suggest on cheking it out and even use his version as it offers many more functions including non-volatile storage and a better low power management https://github.com/pvvx/ATC_MiThermometer`

elalemanyo, 2021-09-27:
thanks @devzwf, after restarting both devices everything worked perfectly.
I already tried the one from @pvvx, but I was having some problems. With both of them I can not connect the device again with the Xiaomi Home app, is this normal?
Thanks

devzwf, 2021-09-27:
well i do not use (and never used) Xiaomi Home app ...... 
but i think if you goona use alt FW it is to get ride of the xiomi app..... am i missing something ?

elalemanyo, 2021-09-27:
> well i do not use (and never used) Xiaomi Home app ......
> but i think if you goona use alt FW it is to get ride of the xiomi app..... am i missing something ?

yes, you are right. The idea was to use it out of Xiaomi app, but just wanted to know if it is possible to have both ;-)

devzwf, 2021-09-27:
sorry cant answer :) as i said never tried

closed 2021-09-27 by devzwf