atc1441/ATC_MiThermometer #145: Advertising interval on Mi-like format

eduperez, 2020-12-27:
When using the Mi-like format, the current model of alternating between temperature+humidity and battery does not make much sense, IMHO.

Battery info does not change that frequently, there is no need to send updates every two cycles. Also, once the device has been awake, it should send both messages at once, to avoid unnecessary wake ups.

atc1441, 2020-12-27:
Hey. 
On mi like advertising it will change the data at half the advertising interval to keep the timing right.

It will not meassure the battery voltage more often so it does not need more power.

In general it will always wake up at an 3 second interval and then always sends the BLE name and advertising data, the interval only changes how often it does change this data.

If it would not send the data every 3 seconds no device would be a ble to connect to it.

Hope that explains it well enough

pvvx, 2020-12-27:
#define ADVERTISING_INTERVAL 3000
https://github.com/atc1441/ATC_MiThermometer/blob/master/ATC_Thermometer/app_config.h#L9
**Advertising interval: 3000 * 0.625 = 1875 ms**

https://github.com/atc1441/ATC_MiThermometer/blob/master/ATC_Thermometer/app.c#L79
Battery measurement interval: 5 * 60000 * CLOCK_SYS_CLOCK_1MS = 5 * 60000 * 24 * 1000 = 0x1AD274800 (overflow 32 bits)
0x1AD274800 -> 32 bit  = 0xAD274800 = 2905032704
Step clock_time() 1/16 us -> 2905032704/16 = 181564544 us = 181.6 sec
**Battery measurement interval: 181.6 sec**