atc1441/ATC_MiThermometer #146: Android apps for reading data from CFW

cricri-pingouin, 2020-12-28:
Hi all,
Not an issue! I have a couple of the Mi thermometers, and currently using the app "Xiaomi Home" to read humidity and temperature. But this app is massively bloated for my purpose.
If I was to flash my units with this CFW, is there an Android app that will let me read temp+humidity data? If not, will "Xiaomi Home" still be able to parse the data from the CFW?
It seems you guys are creating your own custom devices to read data from CFW (e.g. an Arduino + BT shield or something), I just want to carry on getting the data on my mobile phone.
Thanks!

Einstein67, 2020-12-29:
Try "Mija-Temp": https://play.google.com/store/apps/details?id=com.smrtprjcts.mijiabt&hl=de&gl=US

[image]

cricri-pingouin, 2020-12-29:
Awesome, thanks!

ralf-e, 2020-12-29:
Top.

I have just tested with 16 sensors here.
Most of these are LYWSD03MMC with ATC firmware or the ATC power saving trial version of pvvx.
 
Also 2 MHO-C401s, one with the original firmware, one with the e-Ink ATC version from znanev.

In addition 2 pieces LYWSD02 (original firmware because advertising is not encrypted there).

These are queried here with Tasmota (sensor FW) via ESP32 (2 ESP32, because Tasmota currently does not support more than 11 sensors for mqtt without changes / compiling yourself).

The Android app is ideal for a quick overview, especially since the names of the sensors in the displayed list can be adjusted (room name ...)

pvvx, 2021-01-10:
Work "Mija-Temp" with original firmware (Xiaomi). In a month, it will plant a battery.
[image]
[image]
Makes a lot of connection and no adaptation "connection delay"....

pvvx, 2021-01-10:
The Xiaomi Mijia Gateway 3 behaves almost the same, but it establishes connections every 10 minutes or/and when opening Xiaomi apps. The battery with Xiaomi will not last even a year ...

pvvx, 2021-01-10:
Consumption in connection mode for any custom firmware "ATC_xxx":
[image]
(Custom firmware has RF TX Power more than Xiaomi)
There is no difference consumption in connected state and advertising mode. The increase in consumption is only used for starting "negotiation".

smrtprjcts, 2021-01-10:
> Work "Mija-Temp" with original firmware (Xiaomi). In a month, it will plant a battery.
> [image]
> [image]
> Makes a lot of connection and no adaptation "connection delay"....
> 

Hi.
I'm the developer of the app. I had also noticed the too much connections and have fixed it in the v2.4.3.
Thanks for your hard work guys.
Br
SP

pvvx, 2021-01-21:
> I'm the developer of the app. I had also noticed the too much connections and have fixed it in the v2.4.3.

And who will fix the 'Connection latency'?

smrtprjcts, 2021-01-21:
> > I'm the developer of the app. I had also noticed the too much connections and have fixed it in the v2.4.3.
> 
> And who will fix the 'Connection latency'?
> 
> 

Maybe the Google :)) but I doubt it

koostamas, 2024-08-30:
None of the above solutions worked for me, so I created a very simple web app specifically for Android/Chrome (but it might work on other platforms): [ATC Reader](https://atc.koostamas.hu)