atc1441/ATC_MiThermometer #148: 2 Decimals for temp and humidity

gpaizan, 2020-12-29:
Hi and thanks for your work on this firmware, it works great. There is a suggestion I would like to make. Is it possible to advertise both temperature and humidity with 2 decimals? I use the values in some graphs and they look awkward when the variations are low.

atc1441, 2020-12-29:
Hey  
It does not really makes sense to have more resolution as the sensor itself is not as precise.

But look at Victor's repo, he made it so its with more decimals if i am not wrong

cricri-pingouin, 2020-12-29:
> It does not really makes sense to have more resolution as the sensor itself is not as precise.

Totally agree, more than 1 decimal would be meaningless as well below the level of precision of the sensor.
Even if a CFW offered it, it would just make values more difficult to read with no practical benefit.

rjblake, 2021-01-04:
Perhaps only benefit would be so you could calibrate sensors better (i.e. if you have a number of these same sensors). I've made some calibration changes but sometimes tricky to get them aligned. On the other hand, a 0.1 temp difference does not really make a big difference in the greater scheme of things

pvvx, 2021-01-06:
**Advertising in 4 formats:**

1. UUID 0x181A - size 16: atc1441 format 
2. UUID 0x181A - size 19: custom - temperature x0.01C, humidity x0.01%, battery charge level 0..100%, battery voltage in mV, GPIO-pin flags (mark “reset”) and triggers.
3. UUID 0xFE95 - 0x0A: Xiaomi - battery charge level 0..100%, battery voltage in mV
4. UUID 0xFE95 - 0x0D: Xiaomi - temperature x0.1C, humidity x0.1%

 ++ Сonfiguring mode of transferring everything in turn 

**In Connection mode:**

- Characteristic UUID 0x2A1F - Notify temperature x0.1C
- Characteristic UUID 0x2A6E - Notify temperature x0.01C
- Characteristic UUID 0x2A6F - Notify about humidity x0.01%
- Characteristic UUID 0x2A19 - Notify the battery charge level 0..99%
- Characteristic UUID 0x2803 - Notify, frame id 0x33 (сonfiguring or making a request): temperature x0.01C, humidity x0.01%, battery charge level 0..100%, battery voltage in mV, GPIO-pin flags and triggers.

https://github.com/pvvx/ATC_MiThermometer

cricri-pingouin, 2021-01-06:
@pvvx Nice, does your build also include your suggested low power mods?

cricri-pingouin, 2021-01-07:
According to the datasheet that @pvvx dug out and shared here:
https://github.com/atc1441/ATC_MiThermometer/issues/149#issuecomment-756142514

"The sensor covers a humidity measurement range of 0 to 100% RH and a temperature measurement range of -40°C to 125°C with a typical accuracy of ±2% RH and ±0.2°C."

Which confirms that adding any decimals on humidity, or 2 decimals on temperature, would be completely pointless.
If one wants to do so anyway for the kicks, you don't even need to change the advertising format: just add random decimals to the numbers you display!

pvvx, 2021-01-07:
> @pvvx Отлично , в вашу сборку также включены предложенные вами моды с низким энергопотреблением?

Yes. In default settings advertising power consumption ~ 15.5 uA
Advertising interval 2.5 sec, all measures interval 10 sec, 4 types of ad format are transmitted in turn
 (4x2.5 = cycle 10 seconds between measurements)
All intervals are configurable, including low consumption mode for sensor and show modes for LCD display.
If the sensor is in a low power mode, the measurement accuracy is lower, but the total consumption is also even lower.

SzczepanLeon, 2021-01-07:
How to configure intervals? Do you have customized web flasher?

cricri-pingouin, 2021-01-07:
> 
> 
> How to configure intervals? Do you have customized web flasher?

Yes, you can change the advertising interval from the Telink Flasher:
https://atc1441.github.io/TelinkFlasher.html

atc1441, 2021-01-07:
@SzczepanLeon 
https://github.com/atc1441/ATC_MiThermometer#victor-pvvx-did-some-very-nice-refinigs-of-this-custom-firmare-so-i-really-suggest-on-cheking-it-out-and-even-use-his-version-as-it-offers-many-more-functions-including-non-volatile-storage-and-a-better-low-power-management-httpsgithubcompvvxatc_mithermometer

pvvx, 2021-01-07:
> Which confirms that adding any decimals on humidity, or 2 decimals on temperature, would be completely pointless.
> If one wants to do so anyway for the kicks, you don't even need to change the advertising format: just add random decimals to the numbers you display!

Typical accuracy is ± 2% RH and ± 0.2 ° C for sensor in “low power” mode.

pvvx, 2021-01-07:
Tests i2c sensors:
https://wiki.liutyi.info/display/ARDUINO/Test+1+v8b+Balcony+high+to+mid+RH
...