atc1441/ATC_MiThermometer #149: Temperature below 0° Celsius ?

fulvioalessio, 2020-12-30:
I'd like to know if internal sensor can obtain a better range of misure:

in sensor.c I read:
`*temp = ((1750*(read_buff[0]<<8 | read_buff[1]))>>16)-450;`

Which is the range of degrees for 16 bits ?

TIA

atc1441, 2020-12-30:
Hey. It can be up to -19.9°C it is an int16_t variable you need to divide it by 10

fulvioalessio, 2020-12-30:
Hi!
So, if I understand correctly, (65535 / 10) / 100 -> ~ 65° of range ?

thank you

atc1441, 2020-12-30:
You have a range of -32,768 to 32,767 and 15 would be 1.5°C so maximum is + 3276,7°C and - 3276,8°C

:)

cricri-pingouin, 2020-12-30:
That's sub optimal, temperature can't be below -273.15°C, but it could be above 3276.8°C, so you could shift the value to increase the range! But then I'm not sure how well the sensor would perform beyond 3000°C ;)

pvvx, 2021-01-07:
_The sensor covers a humidity measurement range of 0 to 100% RH and a temperature measurement range of -40°C to 125°C with a typical accuracy of ±2% RH and ±0.2°C._
https://www.sensirion.com/en/environmental-sensors/humidity-sensors/digital-humidity-sensor-shtc3-our-new-standard-for-consumer-electronics/

pvvx, 2021-01-07:
LCD show:
Large number: int16 / 10 = -99.5..1999.5 Show 3 ranges: -99 .. - 9.9 .. 199.9 .. 1999
Small number: -9..99
If below - show 'Lo', if above - show 'hi'