atc1441/ATC_MiThermometer #150: BLE GATT UUID for Temperature?

plambrechtsen, 2020-12-31:
Hi

I was using a BLE scanning app to monitor the custom firmware and it didn't detect the temp correctly.
It seems the UUID assigned is 2A1F but I think it should either be 2A1C and specify if it is C or F, or 2A6E for C.
https://github.com/atc1441/ATC_MiThermometer/blob/772348a2266ea22bd58795314be148a8154e6462/ATC_Thermometer/app_att.c#L67

As per:
https://btprodspecificationrefs.blob.core.windows.net/assigned-values/16-bit%20UUID%20Numbers%20Document.pdf
Found via the assigned numbers: https://www.bluetooth.com/specifications/assigned-numbers/

2A1C, with additional values to determine if C or F.
https://github.com/oesmith/gatt-xml/blob/master/org.bluetooth.characteristic.temperature_measurement.xml

2A6E, for just sending C.
https://github.com/oesmith/gatt-xml/blob/master/org.bluetooth.characteristic.temperature.xml

pvvx, 2021-01-03:
**2A1F** - sint16 temperature in 0.1 °C
**2A6E** - sint16 temperature in 0.01 °C
**2A1C** - float

nRFConnect:
[image]
[image]
https://github.com/pvvx/ATC_MiThermometer

plambrechtsen, 2021-01-04:
> **2A1F** - sint16 temperature in 0.1 °C
> **2A6E** - sint16 temperature in 0.01 °C
> **2A1C** - float
> 
> nRFConnect:
> [image]
> [image]
> https://github.com/pvvx/ATC_MiThermometer

Nice one @pvvx 👍