atc1441/ATC_MiThermometer #156: sending data too often?

spants, 2021-01-03:
I have set up ESPhome on esp32 devices to send the received data to mqtt.
I tried with the custom firmware and compiling my own, but it seems that I do not understand how the Advertising Interval should work.
With custom and complied firmware, I have set Interval to 5min but I receive data randomly (
4min/20secs/40sec/80sec) for battery, temperature and humidity updates. 

Is this normal?
Many thanks

atc1441, 2021-01-03:
hey, there is also a threshold on which the data will be notified if the °C or % changes to high

maybe that is what you see ? https://github.com/atc1441/ATC_MiThermometer#temp-or-humi-instant-advertising

eblot, 2021-01-03:
I'm observing the same issue.
Using the custom FW (downloaded binary), after selecting 1 minute delay, I observe ~ 1860 ms interval on both an Android device (using nRFConnect) and Linux x86_64 PC...

atc1441, 2021-01-03:
The 1860ms period is like it should be. 

That is the BLE advertising that is running all the time, only the packet counter will increase on set interval.

If it would not send put these ble packets no device would be able to connect to it anymore, it will be transmitted together with the ble name

eblot, 2021-01-03:
Ok, so it is just a definition issue. What is called the advertising period in the web interface is therefore the sampling period, not the real advertising period, if I understand your reply.

I think it should be possible to reduce the advertising interval (that is, the delay between TX radio is on on BLE channels 37-39) to preserve battery life. It would take longer to be able to connect for sure, but it would not prevent to connect as long as the BLE advertising delay is in the valid range.

Time to try to tweak the FW I guess :-)

Thanks for your answer.

atc1441, 2021-01-03:
Yes that is how it is :) 

In the stock firmware it is called advertising interval so it was taken from there ...

rtoma, 2021-01-03:
What about using this?

```
ble_sts_t 	bls_ll_setAdvInterval(u16 intervalMin, u16 intervalMax);
```
Found in the SDK at https://github.com/Ai-Thinker-Open/Telink_825X_SDK/blob/master/components/stack/ble/ll/ll_adv.h

Same header file that defines `bls_ll_setAdvData`, which is used by this custom firmware.

Bluetooth 4.2 spec says the interval should be a multiple of 0.625ms between 20 ms
to 10.24 s. The observed 1860ms seems to indicate there is potential to save battery life. Not ready yet to verify myself, so maybe someone else can check?

(Thanks @atc1441 for your awesome work. Would love to read how you reverse engineered the stock firmware.)

michapr, 2021-01-04:
it is as atc1441 said - every BLE device need a defined advertising interval - and it must not be very long.

As sample some Gateways are scanning only for 10 seconds every 30 seconds and will not find the device - or will not detect the rare advertising packet.
All other (known to me) BLE sensors without needed connect also reporting the values within such a short interval by default.

So I think the right way would be to modify the ESPhome (or other) software so that it will report only changed values or values after any timeframe.

For battery life time it is not very important how to setup the advertising time as was shown at other place - 18 months will be possible with this settings.

Anyway it is of course possible to compile the firmware with other advertising timeframe for personal use ;)

eblot, 2021-01-04:
It really depends on how you plan to use the devices:
As I use a dedicated (bleson based) BLE central, I do not mind having long advertising period. If possible, I prefer to boost BLE TX (it seems to be 3 dBm (`RF_POWER_P3p01dBm`), and seems to support up to 10dBm) and decrease advertising packets for the same battery budget. Once set up, connecting to the BLE peripheral il only required to upgrade it or change its settings, so long delays should not be an issue. Will see :-)

pvvx, 2021-01-04:
> If possible, I prefer to boost BLE TX (it seems to be 3 dBm (`RF_POWER_P3p01dBm`), and seems to support up to 10dBm) and decrease advertising packets for the same battery budget. 

Datasheet for Telink TLSR8251:
---
**2.6.4 VBAT and VANT power-supply mode**
The chip provides two power-supply modes including VBAT mode and VANT mode.
>* In VBAT mode, the chip is directly supplied with power by its battery voltage. The maximum output power is related to supply voltage, for example, the maximum 10dBm or so at 3.3V power supply.
>* In VANT mode, the chip is supplied with 1.2V voltage by embedded DCDC and LDO. In this mode, output power won’t change with AVDD basically, and the maximum power is 5dBm or so. Corresponding to the VBAT mode, VANT mode is more power-saving at the same Tx power.
---

Maximum VBAT/VANT in SDK: 
VBAT RF_POWER_P10p46dBm = 63  = 10.46 dbm
VANT RF_POWER_P3p01dBm  = BIT(7) | 63   =   3.01 dbm

In VBAT mode, the real TX is below VANT...

pvvx, 2021-01-04:
All formats in turn. 4 in 1.
4 advertising intervals of 2.5 seconds, each with its own format:
2 types 181A and 2 types FE95.
10 sec update of all measurements.

---
[image]
---

Average consumption: 
If sensor in "Normal mode" - 14.9..15.5 uA
In "LowPower mode" - 14..14.5 uA
No longer intervals required...

eisengrau, 2021-01-07:
What would be the best setting in this case, for the best power saving/least updating interval in case of an ESPhome/MQTT/HomeAssitant setup? Let's be honest, everybody who has this tihing will aim for the same: the longest battery life possible. As we are humans, I don't think that any sane person would check the temperature twice per minute or so... :). Also, it would be great that we would have a setting to turn off the display and/or show it every mintue for 5 second then turn off again, or such.

pvvx, 2021-01-12:
> setting to turn off the display and / or display it every time

Will not reduce consumption by 0.1%
Energy is only consumed when the display is updated. If the values do not change, the display is not updated.

If the sensor is in the "low power consumption mode" - the measurement procedure does not exceed the consumption by 1%.

Increasing the ad interval by more than 2.5 seconds will complicate the connection to the device.

metronidazole, 2021-01-22:
@pvvx How much energy is used for each advertisement or measurement? How does compare to the energy used by the display?

eblot, 2021-01-23:
FWIW, I made some measurements using a NRF Power Profiler Kit, powering up the device with 3.25v:

* TX over 1 minute timeframe
  [image]
* Unusual burst, rarely seen, I do not know what triggers this sequence. It could be an ongoing active scan, but that's a guess.
  [image]
* Typical sequence which occurs every 10 seconds w/ "Advertising interval" set to 10 seconds (which means sampling interval)
  [image]
* Details of the first part:
  [image]
* Details of the second part:
  [image]
* Details of the third part:
  [image]
* Single TX every 2.5 seconds:
  [image]

LCD consumption is not measurable (with this tool, but anyway neglectable)