atc1441/ATC_MiThermometer #159: Cannot connect after cf flash of LYWSD03MMC

setdetnet, 2021-01-09:
Cannot connect after custom flash.

Before flash, the sensor is shown in bluetooth device scan  on Samsung Phone, Sony laptop or Raspberry Pi 3. I can also pair and connect the sensor using any of these devices.

Then I flash OTA using the .BIN file in your latest build. After flash, all done well, battery removed, replaced, then sensor. Display reads Atc and then goes through the MAC bytes. So I know its flashed.

However, trying to now scan, pair and connect via linux debian buster bluetoothctl or hcitool no good. Does not show up in Samsung phone bluetooth lists.  The sensor does not show up on a scan. Before flash, it showed up and I can connect using Xiaomi app. After flash, the sensor not seen in bluetooth on Samsung Phone, Sony laptop or Raspberry Pi 3.

Flash back to stock and all is good again.

Reason I want stock is for battery display and battery sensor in Home Assistant.

So now I have to go without battery sensor. But your OTA flash web script does obtain the encryption key so I use that in Home Assistant and it works fine using Xiaomi OEM firmware.

I repeated the custom flash -> stock flash -> custom flash , three times, same result. I do notice that in your OTA web flash , after your custom firmware is flashed, your OTA web cannot connect. It says "NetworkError: Connection Failed for unknown Reason. Reconnect 1 from 5" Then it goes to "Reconnect 5 from 5" then disconnects.

Something definitely wrong.

atc1441, 2021-01-09:
Hey, please try out Victors version of the firmware: 
https://github.com/pvvx/ATC_MiThermometer

the edited web flasher is also available on his repo:
https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html

setdetnet, 2021-01-09:
Hey atc, thanks, but I get same problem, using new OTA and new firmware as recommended, but this time the log box contains just a little more info. It says

"Device Version 1.1, Custom config:[3, 0, 0, 0, 40, 4, 191, 124, 49"

..that line was not in the OTA log box

Flashed back OEM firmware, all good again.

atc1441, 2021-01-09:
what i do not understand is how can you reflash the thermometer if it is not available on a scan ? 

how do you meant that?

setdetnet, 2021-01-09:
I use the web OTA flash. They never appear on BT scan when OEM firmware. I have just noticed, that in Home Assistant, it has already created battery sensors using "discovery" with the Passive_ble_monitor" custom component integration. I was only flashing these with custom firmware to get the battery sensor - but it already has them with stock OEM firmware. They must have changed the firmware.  I have only just bought these today.

setdetnet, 2021-01-09:
to be clear:-

1. With custom f/w , the sensors do not appear on BT scan. On any of my devices/computers/phones.
2. With stock f/w, they appear on a BT scan.
3. When using my phone, and the OTA flash tool, click "Connect" , they all show up when either stock or OEM f/w.
4. With custom yours or pvvx , there is NO connection whatsoever on any device except my phone. They only appear on the web OTA tool on my phone. Scans on other machines do not find the sensor. Flash back OEM f/w and they can be seen on any scan on any machine:

So

Is the firmware using BT protocol or standard that is not on some machines? That's all I can think of.

pvvx, 2021-01-10:
Older and cheap BLE modules do not support long intervals between beacons, advanced settings are not included in the  connection mode and have a short communication distance (low RF TX power). View BT5.0 standards and compare with BT 4.0 ...
Apple devices are even worse than Android - https://github.com/atc1441/ATC_MiThermometer/issues/137

setdetnet, 2021-01-17:
> Older and cheap BLE modules do not support long intervals between beacons, advanced settings are not included in the connection mode and have a short communication distance (low RF TX power). View BT5.0 standards and compare with BT 4.0 ...
> Apple devices are even worse than Android - #137

I don't know any linux OS that supports BT5.0. 

My linux kernel is 5.4.72 with BT 4.1. I am flashing the custom fw to Xiaomi LYWSD03MMC sensor. Does/should this custom fw work with 4.1/my controller and sensor? It is not showing up on a BT scan with your fw in. Reflash to OEM and it's fine.

`hci0:   Type: Primary  Bus: UART
        BD Address: ?:?:?:?:?:? ACL MTU: 1021:8  SCO MTU: 64:1
        UP RUNNING PSCAN
        RX bytes:544799575 acl:119 sco:9 events:13752362 errors:0
        TX bytes:385708 acl:106 sco:0 commands:59248 errors:0
        Features: 0xbf 0xfe 0xcf 0xfe 0xdb 0xff 0x7b 0x87
        Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV2 HV3
        Link policy: RSWITCH SNIFF
        Link mode: SLAVE ACCEPT
        Name: 'Pi'
        Class: 0x000000
        Service Classes: Unspecified
        Device Class: Miscellaneous,
        HCI Version: 4.1 (0x7)  Revision: 0x1fc
        LMP Version: 4.1 (0x7)  Subversion: 0x2209
        Manufacturer: Broadcom Corporation (15)
`

IvxLars, 2021-01-18:
@setdetnet it may change name to ATC_xxxxxx, at least mine did

almostlunatic, 2021-01-18:
Maybe it's not your case, but replace the battery with a new one and know that every new battery isn't necessarily good with a load. It was the reason why my sensors didn't reconnect after the custom fw flash, though they showed 3.1-3.2 volts without a load.

RoganDawes, 2021-01-20:
I experienced the same thing. Flashed with custom firmware (latest release at time of posting), and then could not reconnect to the device from my (Linux, Fedora 32) laptop. Had to use an Android tablet to connect to change various properties.

brianewman, 2021-02-19:
> I experienced the same thing. Flashed with custom firmware (latest release at time of posting), and then could not reconnect to the device from my (Linux, Fedora 32) laptop. Had to use an Android tablet to connect to change various properties.

Thanks for this.  I came here because after flashing the custom firmware, the web flasher could no longer connect to the sensor.    I tried from two different Chromebooks.  After seeing your comment, I tried from my Android phone and it works perfectly.

monte-monte, 2022-01-23:
I have 4 sensors one of them is flashed with custom firmware. All of them are showing on my phone, but when I try to search them with my laptop it only finds 3 that are on stock firmware. To be clear, the one in question is sitting just next to my laptop, but no sign of it in ``bluetoothctl``.

pvvx, 2022-01-23:
@monte-monte 
The [old version of bluetoothctl](https://github.com/pvvx/ATC_MiThermometer/issues/172) has a lot of bugs and glitches with BLE. In older kernel versions, incorrect timeouts are inserted and Bluetooth LE devices with an advertising period according to the https://www.bluetooth.com/ specification are not served.

pvvx, 2022-01-23:
In order for a BLE device to be displayed in the system menus of some operating systems, it must transmitt special flags in the advertising package. This is in addition to the transfer of 3 bytes, which increases battery consumption. But none of the operating systems has special drivers for working with specific BLE devices.
As a result, displaying devices in the system menu to connect to missing system drivers only makes it possible to observe that the device is in the field of view.
The absence of flags for the BLE implementation means that the default flags are accepted, which is not implemented in some operating systems, in their BLE driver.

---

[Alternative version.](https://github.com/pvvx/ATC_MiThermometer)
If you want everyone around you to see your device in the system menus, enable the transmission of special flags:
[image]

monte-monte, 2022-01-23:
@pvvx well, it seems that there is simple solution for that. Thanks for pointing to the thread.
As @skgsergio found out, that's only matter of configuration:
https://github.com/pvvx/ATC_MiThermometer/issues/172#issuecomment-1009670452