atc1441/ATC_MiThermometer #160: problem after telinkflasher to pair again with xiaomi app

walkercz, 2021-01-15:
Hello,

I have Xiaomi Mi Temperature and Humidity Monitor 2 - LYWSD03MMC 

and I am using integration https://github.com/custom-components/ble_monitor

and I used this method to get bind key https://atc1441.github.io/TelinkFlasher.html

problem is that now when i try to factory reset and pair again with xiaomi app ( gnd+reset ) I still get an error -29 :slight_frown: .. anyone who solve it?? and sucesfully repair the thermometer with xiaomi app? (I want to use thermometer in xiaomi app and HA so i need to pair it and get the bind key from modded xiaomi app)

walkercz, 2021-01-15:
I tried to flash stock firmware using telinkflasher - still got an error "Standard verification failed -29"

atc1441, 2021-01-15:
You need to delete the thermometer in the app and re add it.

If that does not solve it try reflashing the newest Custom Firmware and then back to stock

walkercz, 2021-01-15:
Great flash latest custom fw then stock fw worked! Thermometer is connected back in xiaomi app. Thank you for help!

atc1441, 2021-01-15:
No problem 👍

Xiaomi locked the custom firmware id from the cloud so i made an update that will erase that id from flash to be able to use the stock firmware again

sufficientpickle540, 2021-01-16:
Hey @atc1441 or @pvvx, does that version check mean you cannot pair custom firmware with the Mi Home app? I see that it works with MijiaTemp but it gives an error when pairing in Mi Home (assume the key exchange is the issue?).

pvvx, 2021-01-16:
MiHome uses proprietary protocols for connection. 

Telink Mijia example https://yadi.sk/d/g5fV7WD1EaUdRQ
Requires the "libmijia_std_authen_eclipse_telink_1_0_1.a" library from Xiaomi, obtained by subscription:
Connect to Mijia -> https://iot.mi.com/new/doc/embedded-development/ble/standard.html

pvvx, 2021-01-16:
> (I want to use thermometer in xiaomi app and HA so i need to pair it and get the bind key from modded xiaomi app)

Reading and recovering Mi-Home keys
https://github.com/pvvx/ATC_MiThermometer/tree/master/img

gitkobold, 2021-03-13:
Hi.
Are only the lywds03 and mho-c401 use telink (and supported by the page)?
I have more types and as i see there are lots of other xiaomi sensors and it would be good to flash original binaries (w/o mi home,) at least..
Thanks

lolax80, 2021-04-13:
Can somebody Tell can i bind the Customer firmware with the Mia Home App on iOS?
I dont Unterstand Reading and recovering Mi-Home keys.

Stratos007, 2021-05-05:
> > (I want to use thermometer in xiaomi app and HA so i need to pair it and get the bind key from modded xiaomi app)
> 
> Reading and recovering Mi-Home keys
> https://github.com/pvvx/ATC_MiThermometer/tree/master/img

Unfortunately, it didn't work for me.

Stooovie, 2021-06-04:
After flashing custom,  I can't Do activation anymore (so, I can't get further). After pressing Do Activation, it always disconnects. I can't flash back to stock and re-add to Mi Home.

atc1441, 2021-06-04:
Hey.
You only need to Activate on stock firmware.

On custom firmware you can directly flash after connecting

Stooovie, 2021-06-04:
> Hey.
> You only need to Activate on stock firmware.
> 
> On custom firmware you can directly flash after connecting

no I can't, the option is not there. The custom FW (button straight from the flasher) completed succesfully, no error messages.
[image]

atc1441, 2021-06-04:
Looks like you are using pvvx version. Use the one from this repo please

Stooovie, 2021-06-04:
How if I can't flash anything? I'll probably need to do an UART unbricking, right? Thanks for your time.

atc1441, 2021-06-04:
No, i am talking about the telink flasher from this repo and not pvvx ones :)

https://atc1441.github.io/TelinkFlasher.html

Stooovie, 2021-06-04:
Yes, sorry, thanks :) I have used your flasher to flash the original firmware you link to, it seems to have gone OK, but I can't add the device in the MI Home app no matter what I try. I did do the hardware reset (short gnd and reset for 5-8s) already. Any ideas? Thanks!

atc1441, 2021-06-04:
Please delete it from the App first and activate it again.

Also before that flash my (not Pvvx) firmware once and then the stock again. This will erase the Old MiKeys.

Stooovie, 2021-06-04:
Thanks, I was able to flash stock by repeating the process multiple times.

pvvx, 2021-06-04:
> Also before that flash my (not Pvvx) firmware once and then the stock again. This will erase the Old MiKeys.

[image]

jpdribbler, 2022-06-03:
> > Also before that flash my (not Pvvx) firmware once and then the stock again. This will erase the Old MiKeys.
> 
> [image]

Thanks. This method fixed it for me. Now i am able to pair the device with mihome again.

SwingCZ, 2023-11-11:
> Great flash latest custom fw then stock fw worked! Thermometer is connected back in xiaomi app. Thank you for help!

Hello, I am in the same situation - I have updated my thermometer via Mi Home to FW original version 2.1.1_0159, but after this, the was not possible to read data from device via Mi Home. I deleted this device from Mi Home and tried to pair again - but there is not possible, notice "standard verification failed (-29)" is still showing.

Please, any tips how to fix it?
Thanks.

Alfred99-HA, 2023-11-22:
https://atc1441.github.io/TelinkFlasher.html "Do activation" fails withoput any error message, it just remains in Status
	16:00:10: Activating now, please wait...
and thus flashing another firmwar is not possible.

https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html at least reports:
	16:04:16: Detected Mi device
	16:04:16: (!) Version: 2.1.1_0159 is temporarily not supported (!)
	16:04:16: Disconnected.

I hope version  2.1.1_0159 will be supported soon - the only task I would like to do, is to flashe another firmware