atc1441/ATC_MiThermometer #162: Time value overflow?

eblot, 2021-01-17:
app_config.h:7 `CLOCK_SYS_CLOCK_HZ = 24000000`
app.c:79:  `5*60000*CLOCK_SYS_CLOCK_1MS`
```
CLOCK_SYS_CLOCK_1S = CLOCK_SYS_CLOCK_HZ
CLOCK_SYS_CLOCK_1MS = CLOCK_SYS_CLOCK_1S / 1000
5*60000*CLOCK_SYS_CLOCK_1MS= 7.2e9
```
-> `5*60000*CLOCK_SYS_CLOCK_1MS` > `UINT32_MAX`

atc1441, 2021-01-17:
Hey. 

Yes that is a known issue.

Please take a look at @pvvx firmware edition, it is way more advanced now :)

pvvx, 2021-01-18:
https://github.com/atc1441/ATC_MiThermometer/issues/145#issuecomment-751494015

giannoug, 2021-01-19:
@atc1441 So is this project abandonded? Should we migrate to https://github.com/pvvx/ATC_MiThermometer?

atc1441, 2021-01-19:
Right now it is abandonded yes,
I would not like to simply copy all the work from @pvvx as he improved so much its more grateful to directly use his fork :)