atc1441/ATC_MiThermometer #163: Advertising Messages

PaulMcGuinness, 2021-01-20:
Hi - I'm pretty new to these, so sorry if it's a dumb question...  I am running a BLE advertising monitor app (the example Ardiuno one), but I never see any temp/humidity messages, only a standard advertising message.

```
Advertised Device: Name: , Address: 59:64:29:32:7a:3d, manufacturer data: 0600010920021c014e3814bbd9938dec5db86d48c55adb022bd188f3be 
Advertised Device: Name: ATC_03F4A2, Address: a4:c1:38:03:f4:a2 
Advertised Device: Name: , Address: 55:50:a2:0b:a1:5d, serviceUUID: 0000fd6f-0000-1000-8000-00805f9b34fb 
Advertised Device: Name: ATC_2941BD, Address: a4:c1:38:29:41:bd 
Advertised Device: Name: ATC_120632, Address: a4:c1:38:12:06:32 
Advertised Device: Name: ATC_665564, Address: a4:c1:38:66:55:64 
Devices found: 6
Scan done!
```

I have 4 x temp sensors, all running either the 1.2 or 1.4 firmware, and although I can see "data" messages from something(?), I'm only see MAC address info for the temp sensors.

Any idea?

Cheers,

Paul

pvvx, 2021-01-21:
There are no standards for advertising messages.
Typical formats for BLE beacons only.

PaulMcGuinness, 2021-01-21:
Yes, I get that.... The point I was making is that there doesn't seem to be anything broadcast except the MAC address, or am I missing something.

sthex, 2021-02-20:
I just tried the same example. You have to ask for the payload to get the intresting data.
    void onResult(BLEAdvertisedDevice advertisedDevice) {
      Serial.printf("Advertised Device: %s \n", advertisedDevice.toString().c_str());
      **uint8_t* payload = advertisedDevice.getPayload();
      size_t payloadLength = advertisedDevice.getPayloadLength();**

Take a look at this repo: https://github.com/AlexNS/xiaomi-temperature-mqtt-example

PaulMcGuinness, 2021-02-24:
@sthex - Cheers! Exactly what I was looking for :-)

closed 2021-02-24 by PaulMcGuinness