atc1441/ATC_MiThermometer #164: Block/Secure ota flashing

podarok, 2021-01-21:
Hi.
Thank you for your work on this great piece of software

Is there any way to lock ability to flash firmware?
Seems like anyone can flash by having a mobile phone and opening ota flasher in browser

atc1441, 2021-01-21:
Hey.

This could be added to the custom firmware yes.

There was already a discussion about possible ways to do it

pvvx, 2021-01-21:
Set pin code + binding. Works since version 1.2. https://github.com/pvvx/ATC_MiThermometer
If you forget the pin code, then only hardware firmware... Erase all flash.

atc1441, 2021-01-21:
That is perfect!

Thank you Victor

podarok, 2021-01-21:
@pvvx is there any documentation on how to set pin code?

pvvx, 2021-01-21:
Most html elements have a tooltip on the page:
[image]

pvvx, 2021-01-21:
[image]

pvvx, 2021-01-21:
Windows 10 has a specially crafted security hole. When Windows 10 detects a similar device, but with different MAC addresses, you will be prompted for a pin code for your device. This happens a few minutes after connecting to your device, so you yourself enter your PIN to the attacker. After restarting drivers and hardware, these prompts will disappear.

pvvx, 2021-01-21:
After the bind, on Windows 10, the Battery icon shows last year's information.

podarok, 2021-01-21:
@pvvx I tried to set a pin code. Next time reconnecting to a device from https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html it shows a modal from OSX to start pairing mode or to cancel  it. I'm cancelling and without pin code I can reflash device. 
So I still have no idea how to lock device from flashing by anybody

podarok, 2021-01-21:
[image]

pvvx, 2021-01-21:
> So I still have no idea how to lock device from flashing by anybody

Delete the paired device.
[image]

'''
Log:
00:56:05: Searching for devices
00:56:33: Connecting to: ATC_0F251C
00:57:06: Detected custom Firmware
00:57:06: Hardware Version: LYWSD03MMC.
00:57:06: Software Version: 1.4, Custom config: [39, 0, 0, 0, 40, 4, 191, 124, 49]
00:57:27: File was selected, size: 52148 bytes
00:57:27: Count: 3260
00:57:32: Start DFU
00:57:42: Update error: NotSupportedError: GATT operation failed for unknown reason.
00:57:42: Disconnected.
00:57:58: Searching for devices
00:58:16: Connecting to: ATC_0F251C
00:58:25: Disconnected.
'''

Android - the reaction is the same: disconnected.

Binding, PIN-code is remembered by the communication device (adapter) - hard reboot.

pvvx, 2021-01-21:
I'll see what else can be inserted to ignore the wrong or missing pin-code...

pvvx, 2021-01-22:
@podarok  
pincode = "123456" ?   :) :) :)
Default pincode = '000000' & '123456'.

podarok, 2021-01-25:
Hi @pvvx 
No, pincode is tough, all good there
Accordingly to your suggestion "Delete the paired device." - frankly speaking - these devices are not in the list of paired in OSX as well as Android I was using to flash custom firmware

I've set all PIN codes and right after setting Both Android and OSX asked me to provide a PIN which I declined. Web flasher still was able to reach out settings and firmware update

But once I removed connected ATC from the list
[image]

And retried to connect after that - no luck
So the trick is - to remove a device from paired after setting a PIN code in Flasher

podarok, 2021-01-25:
Nope
Here is a proof
After removing the device from a list of paired and reconnection from flasher with Cancelling pairing process still allows you to interact with firmware
https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html
Proof https://youtu.be/tuupn1XYASA

podarok, 2021-01-25:
And flash updated with no issues 
[image]

atc1441, 2021-01-25:
Maybe using a propriatary pin method is simpler in this case

podarok, 2021-01-25:
> Maybe using a propriatary pin method is simpler in this case

which one are you referring to?

atc1441, 2021-01-25:
A fully custom one, a flag that will be set and enables all cmds only after the correct pin is received via the controll uuid that is also used for the settings

podarok, 2021-01-25:
Ah, got it @atc1441 
I don't think I have the capacity to work on this kind of development
Looking forward to seeing this implemented some day

pvvx, 2021-01-25:
> Maybe using a propriatary pin method is simpler in this case

PIN, binding, pairing are stored in BT adapter or system registry. Until they are removed from there.
The PIN code "000000" and / or "123456" is sent by the system by default.
The standard SMP method is used.

pvvx, 2021-01-25:
[image]

pvvx, 2021-01-25:
> Нет.
> Вот доказательство.
> После удаления устройства из списка сопряженных и повторного подключения из флешера с отменой процесса сопряжения вы все еще можете взаимодействовать с прошивкой
> https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html
> Proof [https: // youtu .be / tuupn1XYASA](https://youtu.be/tuupn1XYASA)

How is a keyless encrypted connection? :)

pvvx, 2021-01-25:
[image]

podarok youtube:

[image]

:) :) :)

podarok, 2021-01-25:
Hi, @pvvx
I've tested flasher on completely new Android phone, which was never used for connecting Bluetooth in my home. And here is how to break pin protection

- open web flasher
- search for ATC_ device with PIN protection
- try to connect to it
- when system opens a modal window with pairing procedure - hit switch window and return back to browser.
- woops, but everything accessible now in flasher.
Proof uploading to youtube

podarok, 2021-01-25:
Here is how it works. (Samsung phone)
Same procedure on Xiaomi MI 9T (after removing all Bluetooth cache in order to overcome caching issue, mentioned above)
https://photos.app.goo.gl/9QZAckM4gpQihmiv6

pvvx, 2021-01-26:
> * woops, but everything accessible now in flasher.

Not work - no connect. Tested on 7 devices, 4 operating systems.

pvvx, 2021-01-26:
Windows 10:
[image]

pvvx, 2021-01-27:
> A fully custom one, a flag that will be set and enables all cmds only after the correct pin is received via the controll uuid that is also used for the settings

GATT Services and Characteristics.
Attributes Permissions are metadata that specify which ATT operations can be executed on each particular attribute and with which specific security requirements.

Access Permissions:

- None
- Readable
- Writable
- Readable and writable

Encryption:

- No encryption required
- Unauthenticated encryption required 
- Authenticated encryption required

Authorization

- No authorization required
- Authorization required

Added in firmware version 1.7.

podarok, 2021-02-16:
Seems like it does work
Flashed 2.1 and without pairing + pin I was unable to overcome pin anymore
Thank you!!!!

closed 2021-02-16 by podarok