atc1441/ATC_MiThermometer #167: Offset in my own Custom Firmware

Bibo1984, 2021-02-02:
Hello

i'hd build my own Firmware with some Custom Settings an it works perfect.
Now i've tryed to enter a Offset for each of them....
But it not works, wich format i've to use that it works correctly?

I tryed: humi_offset = 10;
[image]

thanks for your help!

atc1441, 2021-02-02:
Hey, 

the temp_offset can range from -12,6 to 12,5 °C and will add or subtract 0,1°C for each number
the humi_offset can range from -50 to 50 and will add or subtract one percent for each number

hope that this helps

Bibo1984, 2021-02-02:
@atc1441 
Thanks for the fast help! I've found the mistake the problem was not the parameters in the firmware.
The Telink flasher doesn't upload the bin file again till i close and reopen it again.
It works now very fine. Thanks for your great work!

atc1441, 2021-02-02:
Thank you for the feedback :)

closed 2021-02-02 by Bibo1984