atc1441/ATC_MiThermometer #170: Platform support for Telink Flasher?

YvoTuk, 2021-02-05:
Am I right to assume the Telink Flasher only works on Android phones, or at least not iPhones?

I've tried following the tutorial a couple of times, but tapping connect doesn't seem to do anything?

atc1441, 2021-02-05:
Hey. Yes, see this https://github.com/WebBluetoothCG/web-bluetooth/blob/master/implementation-status.md

pvvx, 2021-02-05:
Advertisements Scanning ?
[image]
[image]

YvoTuk, 2021-02-05:
Thank you! Got it running on a Mac with Chrome browser