atc1441/ATC_MiThermometer #174: Zigbee Xiaomi Mijia ?

pvvx, 2021-02-11:
https://item.taobao.com/item.htm?id=630309914025

znanev, 2021-02-11:
Also this one - I was skipping it for almost two years in favour of BLE based ones (before I got into ZigBee stuff really):

https://www.aliexpress.com/item/1005001611948819.html

Now I'm seriosly considering to buy some of those (powered by 2 x AAA batteries is a another big plus for me).

pvvx, 2021-02-11:
ZiпBee consumes more energy per unit of transmitted information.

znanev, 2021-02-12:
> ZiпBee consumes more energy per unit of transmitted information.

That's true, but seamslessly forming mesh networks and usually having better range leans in favour of ZigBee (for me at least, that is). Also the very nature of ZigBee, that it's event driven and hence the device is in deep sleep most of the time, is a nice bonus for drawing less energy from the battery.

That is not to say that BLE doesn't have its merits - in fact all my temperature / humidy sensors are currently BLE-based. But the radiator valves, motion / light / shock sensors that I own, are all ZigBee based.

G1K, 2021-02-20:
we managed to get a memory dump of the zigbee

 thermometer. I tried to fill it via OTA, a stripped-down version of 160 kb, filled only by 82%

it was flashed through the uart and entered the zigbee network. The testimony is given. But the screen flashes all the segments at once and does not show the temperature.
I tried flashing a full dump, but it didn't change the situation. Judging by the firmware, it was written on the 8538 stack, most likely for TuYa , but the model is listed as for compatibility lumi.wheater

Could someone help me figure out the firmware ?

pvvx, 2021-02-21:
> Could someone help me figure out the firmware ?

Where can I see the firmware?

G1K, 2021-02-22:
If it is possible write in telegram.

https://t.me/mrg1k