atc1441/ATC_MiThermometer #175: "Custom firmware found"

AndreScalaPT, 2021-02-23:
Hello, after I had one sensor with the old version and when I try to connect to him he gives a error message "Custom firmware found" and don't connect.

So I took a new one and I flash him with the last version and everything went ok but now if I try to connect to same one he gives the same error!!!

So, do I lost the ability to connect back and update or change for stock firmware ???

Thanks

atc1441, 2021-02-23:
hey, it is already connected when it says "Custom firmware found"  from there on you can directly flash, no need to activate it, only on Stock firmware activation is needed.

please also check out PVVX  version of this firmware, it is way more advanced now
https://github.com/pvvx/ATC_MiThermometer

AndreScalaPT, 2021-02-23:
Ow I see!!
But still, I don't find the option again to choose different version for the firmware.
What I'm missing?
Sorry, still try to learn :P

[image]

atc1441, 2021-02-23:
Please connect with my Telink Flasher version and install the newest pvvx version

AndreScalaPT, 2021-02-23:
Hmm I didn't know that existed a different version for this.
Probably I mixed versions! Oh my!

This is your last release - https://github.com/atc1441/ATC_MiThermometer/releases/tag/34
And your Telink - https://atc1441.github.io/TelinkFlasher.html

I was scratching my head very hard with this.
Thanks for you patient and support ;)