atc1441/ATC_MiThermometer #180: Esp32 Mqtt relay program?

tablatronix, 2021-03-14:
Has someone made an esp32 to mqtt relay using atc firmware broadcasts? I though I saw someone post about one somewhere.. 

Add external resources links etc.

tablatronix, 2021-03-14:
I see esphome is capable but native ble is iffy.
Any standalone libs?

H4R0, 2021-03-15:
I used OpenMQTTGateway.

https://github.com/1technophile/OpenMQTTGateway

https://compatible.openmqttgateway.com/index.php/product/xiaomi-mi-jia-ble-lywsd03mmc-indoor-mini-weather-station/

stevoh6, 2021-05-25:
I have it, working like a charm. Couple of them have stock firmware, couple this custom with "mi like" advertisment. 

Here is part of my config:
`
esp32_ble_tracker:  
sensor:
  - platform: xiaomi_lywsd03mmc
    # name: 'Living room'
    mac_address: "A4:C1:38:xx:xx:xx"
    bindkey: "hjghjghjghjghjghjgh"
    temperature:
      name: "ble2mqtt temperature X"
    humidity:
      name: "ble2mqtt humidity X"
    battery_level:
      name: "ble2mqtt battery X"
  - platform: xiaomi_lywsd03mmc
    # name: 'Kitchen'
    mac_address: "A4:C1:38:xx:xx:xx"
    bindkey: "gjghjghjghjghjghjgh"
    temperature:
      name: "ble2mqtt temperature IX"
    humidity:
      name: "ble2mqtt humidity IX"
    battery_level:
      name: "ble2mqtt battery IX"
`

If you include also mqtt config, it will send values throught mqtt
`
mqtt:
  broker: $mqtt_host
  port: $mqtt_port
  username: $mqtt_user
  password: $mqtt_pass
`

https://esphome.io/components/esp32_ble_tracker.html
https://esphome.io/components/binary_sensor/ble_presence.html#esp32-ble-tracker-setting-up-devices
https://esphome.io/components/sensor/xiaomi_ble.html?highlight=xiaomi_lywsd03mmc

lixas, 2021-08-19:
I have multiple devices at my flat and one ESP32 running to proxy all advertisements to MQTT. Solution is Micropython based.
https://github.com/lixas/SmartHome-Proxy

tablatronix, 2021-08-20:
Thanks for all the suggestions, I have been running openmqttgateway on an esp32, its been working great

unrelentingtech, 2021-09-25:
I've extended the MiBeacon support in https://github.com/shmuelzon/esp32-ble2mqtt to support more devices, including ones that send v4/v5 encrypted advertisements.

fokcuk, 2022-01-04:
Anyone using free public MQTT broker with any of the solutions? Read that ESP32 does not support TLS and I was not able to find a free, reliable one like HiveMQ that does NOT use TLS