atc1441/ATC_MiThermometer #183: Can't read history after flashing custom firmware

Stolyassa, 2021-04-04:
Hello,
When I have official firmware I can't read any data at all because of the encryption. (Mi home app does not work, don't have mi phone or hub)
When I flash the device I can happily read the temperature and humidity for current time using nRFConnect. But I can't subscribe, see or read history data inside the mi device. Tried nRFConnect and 2 other apps. Nothing changed.

Do you have any ideas? 

I had to go back to the official firmware nRFConnect can see the history but its encrypted now.

atc1441, 2021-04-04:
Hey. This firmware simply not support reading of history data.

Check out pvvx firmware version. There it is possible, its linked in the repo

pvvx, 2021-04-07:
> When I have official firmware I can't read any data at all because of the encryption. (Mi home app does not work, don't have mi phone or hub)

Official firmware does not use history encryption

https://github.com/pvvx/ATC_MiThermometer/issues/54

Stolyassa, 2021-04-10:
Its weird because I can't read it with the official firmware. And can't read the history with pvvx firmware either. I only can see the last day.

closed 2021-04-10 by Stolyassa