atc1441/ATC_MiThermometer #185: Issue reverting back to standard firmware and connecting to Mihome app

MTRatcliffe, 2021-04-17:
After flashing a LYWSD03MMC with the ATC_Mithermometer to test it out, I re flashed the stock firmware and attempted to connect it again to the mihome app, but get this error message: 
[image]

MiHome app info:
-running on android google pixel 2xl
-country/server "china mainland" 

Tried so far:
-Re flashed both modified and original version a few times in case of uploading issue
-Deleted the sensor from the mihome app
-Uninstalled and reinstalled the Mihome app

Has anyone had similar issues and would care to point me in the right direction?

atc1441, 2021-04-17:
Hey. Please try put victors version of this firmware.
He has added key handling into it.

It is linked in the readme

MTRatcliffe, 2021-04-18:
Thank you, I will give it a go.

pvvx, 2021-05-27:
1. Much depends on the load on the Xiaomi server. Xiaomi Clouds is always busy.
2. When re-registering, if the old keys are not deleted before the official version is flashed, it may take up to 4 attempts to bind with MiHome.